2dom/PxMatrix #1: Support for all display types

2dom, 2018-03-29:
Continued discussion from https://github.com/2dom/P10_matrix/issues/7

irazaq, 2018-03-30:
tested it with the ESP.wdtFeed() and the "Pixel Time" display hung for about 29secs before it crashed.
After some playing around, I decide to increase the value of the ticker period from 0.001 to 0.002 as I assume it was a timer issue. The problem disappear and now it is working with show_time of 70 without the watchdog feed.

2dom, 2018-03-30:
Great ... no flicker with the new period ?

irazaq, 2018-03-31:
not noticeable.
I changed the delay period in the loop to 10, 500, 1000ms. 
[P5-files.zip](https://github.com/2dom/PxMatrix/files/1865178/P5-files.zip)

2dom, 2018-03-31:
ok ... looking good - great! Will push this change to master.

closed 2018-03-31 by 2dom