2dom/PxMatrix #5: Scanning Issue

bridge2nowhere, 2018-05-07:
I'm getting a weird effect. I purchased [this panel](https://www.aliexpress.com/item/SRY-1-4-Scan-10mm-Pixel-Pitch-LED-Advertising-Digital-Display-Board-Dot-Matrix-32x16-P10/32711913253.html?spm=a2g0s.9042311.0.0.UjdxP5).

I have it wired correctly, I set it for 1/4 scan, and the pixel test does three lines (and skips one) for each scan. Then it does two middle lines, then top, skip, middle, skip. Then some random purples leds scan across.

I've tried 8,16, and 32 but they don't work either. Am I missing something silly?

Here is a video if that helps:
[](https://www.dropbox.com/s/tvim4a4pbq30cc1/IMG_4764.MOV?dl=0)

2dom, 2018-05-07:
That looks very weird indeed ... could you send a high res picture of the back of your matrix including your wiring?

bridge2nowhere, 2018-05-08:
[image]

The pinout, not well pictured is

```
R1||G1
B1||Gnd
R2||G2
B2||Gnd
 A||B
 C||D
CK||LAT
OE||GND
```

2dom, 2018-05-10:
Sorry...cannot see anything on the pic...

bridge2nowhere, 2018-05-10:
I'm sorry, are these better?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

NeoTrace82, 2018-05-10:
got the same pannel, just wanted to write an issue,
seems like the Patterns are missorderd

Kamal-Sonani, 2018-05-11:
[THIS LINK](http://www.instructables.com/id/Tetris-Time-on-a-P10-RGB-Matrix-With-ESP8266/)

I also comment this type of problem 
See pictures and one Video about that.

It is P10 outdoor module which i generally used to make rgb led screen driven by ready-made chinese controller [HD W60 75](http://www.huidu.cn/product/showproduct.php?lang=en&id=41)

2dom, 2018-05-11:
@Kamal-Sonani ... could you please also run the pattern test and send me a video of the result (I need to see at least two red,yellow, white sequences)

2dom, 2018-05-11:
@bridge2nowhere What is really strange about this is that we just shift in one byte after the other but you display lights up 2bytes worth of LEDs. Could you try pulling all A,B,C pins to ground and repeat the pattern test? (I need to see at least two red,yellow, white sequences)

2dom, 2018-05-14:
Great ... I think we might be getting somewhere. It appears that these panels use a different method of row addressing (no multiplexing at all - just pulling on of 4 lines to low). Please pull the new version of the github master (I just updated the library) and put

```
display.begin(4);
display.setMuxPattern(STRAIGHT); 
```

in the setup and let me know if it works (and send a video if it doesn't :)

bridge2nowhere, 2018-05-14:
No change, it's exactly the same as the last video I posted.

2dom, 2018-05-14:
DId you re-attach all the ABCD line to the microcontroller?

JammyDodger231, 2018-05-14:
This is what my panel is doing with the new code in place and ABCD pins connected: https://photos.app.goo.gl/5CFbC3QFbDbslw9Y2

The code I have used is as follows:
`#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setMuxPattern(STRAIGHT); 
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}`

bridge2nowhere, 2018-05-14:
I did forget the ABCD lines, I will test again tomorrow to see if get the same result as @JammyDodger231

JammyDodger231, 2018-05-14:
With the addition of @2dom's new code I get what I see in my video, if I comment out the new line `display.setMuxPattern(STRAIGHT); ` then I get the same initial video as @bridge2nowhere

2dom, 2018-05-14:
Ok ... had a typo in the code ... uploaded a new version. Can you try again?

2dom, 2018-05-14:
Just keep testing if you don't mind ... I think we almost got it ... try 52b3510

JammyDodger231, 2018-05-14:
Of course 👍  Pulling now

2dom, 2018-05-14:
oh ... and you have to put the display.setMuxPattern(STRAIGHT); before display.begin() for now

JammyDodger231, 2018-05-14:
Changes pulled and made the alteration to my sketch so the MuxPattern is before the begin, this is the result now:
https://photos.app.goo.gl/nsJxIOM83vofjmZw1

2dom, 2018-05-14:
looks perfect ... time to try Pixeltime with 

```
void display_updater()
{
//display.displayTestPattern(70);
display.display(50);
}
```

JammyDodger231, 2018-05-14:
It's close, I have added an image to show what I see. Here is my new sketch: https://gist.github.com/JammyDodger231/61c20563749e86912efea85cb425e696
[image]

2dom, 2018-05-14:
ok ... next one ... f00f8a7 is up (this is more a guess what it could be but try anyway :)

JammyDodger231, 2018-05-14:
Looks closer again! :) 
[image]

2dom, 2018-05-14:
could you put a 
```
for (int xx=0, xx<16;xx++);
display.drawPixelRGB888 (xx,xx,128,128,0);
```
in the sketch?

2dom, 2018-05-14:
wow ... wtf is that. I rolled back the last change ... lets try again 8c7858f

JammyDodger231, 2018-05-14:
Almost the same as previous, just shifted over by three it seems:

2dom, 2018-05-14:
hey @JammyDodger231 ... I will have to analyze this a bit. Let's continue another day - thanks for you help!

JammyDodger231, 2018-05-14:
No problem! Thanks for the help so far 😄 Any progress is still progress!

2dom, 2018-05-15:
OK ... I created a new test to expose the pattern in detail. Please pull the new libraray and place

```
void display_updater()
{

  display.displayTestPixel(70);

}
```

in your sketch (and post the result).

JammyDodger231, 2018-05-15:
Changed now, new test gist: https://gist.github.com/JammyDodger231/22bbcc24700a3a2404a3582358a08830 
New video:
https://photos.app.goo.gl/RbL8Qcd5pSFEqfCLA
Two complete cycles

2dom, 2018-05-15:
OK ... could you pull the new version and try it with

```
display.begin(4);
display.setMuxPattern(STRAIGHT); 
display.setScanPattern(ZAGGIZ);
```

then run 

```
for (int xx=0, xx<16;xx++);
display.drawPixelRGB888 (xx,xx,128,128,0);
```

in somewhere.

2dom, 2018-05-15:
oh ... and don't forget to put

```
void display_updater()
{
//display.displayTestPattern(70);
display.display(50);
}
```

in again

JammyDodger231, 2018-05-15:
It looks like your most recent stuff isn't pushed, the last update was 2 hours ago :)

2dom, 2018-05-15:
my apologies ... should be up now

2dom, 2018-05-15:
0bd9e95

JammyDodger231, 2018-05-15:
Cool I will test now :)

2dom, 2018-05-15:
sorry ... fucked some stuff up ... try again cbaf7ef

2dom, 2018-05-15:
try again ? d531df3

JammyDodger231, 2018-05-15:
Awesome! That seems to have done the trick, I don't imagine you need a video or anything but if you do just let me know :) Thank you for taking the time to sort this out 😄 
[image]

2dom, 2018-05-15:
Yay ... success. What are weird ass display! Glad that it works! 
Maybe send me a picture of your project if you're finished ... always curious what other makers are up to :)

JammyDodger231, 2018-05-15:
Definately will do, its a work in progress but will show you what it looks like, most of it will be copied from your tutorial since it looks good :)

closed 2018-05-16 by 2dom

2dom, 2018-05-16:
> Thank you guys for all the hard work

A pleasure...have fun :)