2dom/PxMatrix #10: Support for 1/2 Scan displays?

lgaryfallos, 2018-05-18:
Hi and thanks for this really useful library.

I managed to buy the wrong display [(this one)](https://www.aliexpress.com/item/Lowest-price-outdoor-smd-led-module-p10-pixel-pitch-10mm-16x32-160x320-rgb-matrix-outdoor-diy/32671503435.html?spm=a2g0s.9042311.0.0.71ea4c4d1tIUgS) since I didn't notice the scan pattern restrictions, but I was wondering if it was possible to support 1/2 scan pattern displays.

I tried setting display.begin(2) but that didn't change anything, I still get double lines when running the pattern test.

2dom, 2018-05-18:
Can you post a video/gif of the pattern test. Also, can you post a picture of the input connector?

2dom, 2018-05-27:
Hi,

could you pull the A and B pins to VCC and repeat the pattern test for me. Also, please post a higher frame rate gif or a video. Thanks Dom

lgaryfallos, 2018-05-29:
@2dom sure, I'll try to, since I disconnected the nodemcu and now the arduino IDE can't find the port again.

When you say pull A and B to VCC do you mean connect them from PI on the panel to 3V pins on the mcu? sorry for the questions but I am a complete beginner in this.

lgaryfallos, 2018-05-29:
@2dom hi again,

I managed to connect the nodeMCU to the Arduino IDE but after I upload the test pattern sketch nothing happens.

The connections are done like this:

The PI to PO as described in the readme.
Then:

- STB -> D0
- A -> 3V
- B -> 3V
- OE -> D4
- CLK -> D5
- R1 -> D7

Nothing happens now when I upload the test pattern.
No changes were made on the test pattern sketch (meaning the P_A is defined to 5 and P_B to 4 as it was (if that makes a difference)

lgaryfallos, 2018-05-29:
This is a video of the current pattern test : [https://youtu.be/5FoLHVvssQs](https://youtu.be/5FoLHVvssQs)
And this is with the standard connections (the one shown in the gif in my previous comment) : [https://youtu.be/GYfHvh_qA6k](https://youtu.be/GYfHvh_qA6k)

2dom, 2018-05-29:
Ok so far...now pull A to GND and repeat.

Thanks Dom

lgaryfallos, 2018-05-29:
I did that (A -> GND), I think it's doing the same (I got a video if you want me to upload it), but it's still too faint to be able to make out what it shows.

2dom, 2018-05-29:
Ok...let’s try A and B to GND then

lgaryfallos, 2018-05-29:
Ok, now it show this: [https://youtu.be/G06Ij4N_L2w](https://youtu.be/G06Ij4N_L2w), that is flashing rows (white I think) but with flashes of other dim colors in between.

2dom, 2018-05-29:
Ok...I think I know what’s going on. Will try to  add support for your display in the next few days ... thanks for testing!

2dom, 2018-06-01:
No problem ... try the latest master and let me know if it works :)

mike-rankin, 2018-06-01:
Before posting this issue I grabbed the latest master but had no luck. All the examples I’ve seen use the 64x32 that works perfectly fine.

2dom, 2018-06-01:
Are you talking about this issue or the other one you posted for the 64x64?

I updated the library a few minutes ago to support 1/2 scan on 32x16 displays

2dom, 2018-06-03:
@lgaryfallos Did you test?

lgaryfallos, 2018-06-04:
@2dom I'm trying it out now. I cloned the repo and ran the pattern test.
How should A&B be connected? They are now at GND as of last time.

2dom, 2018-06-04:
Connected to the microcontroller as described on github

lgaryfallos, 2018-06-04:
Ok, I double checked the connections, I deleted the pxMatrix library from the Arduino IDE and reinstalled the new master from zip. I verified that the new readme is in the libraries folder (so I assume I have the latest version that mentions 1/2 displays). 
[This](https://youtu.be/jsK3rd886Js) is a video of the whole process (sketch upload and running until it gets to the faint part). 
And [this](https://youtu.be/QpM1zz7f1RY) is the same but with begin(2).

2dom, 2018-06-04:
Ok ... I would check the supply voltage and reduce the SPI speed wiithin the libraray to 10Mhz

lgaryfallos, 2018-06-04:
Sadly, I have no way of checking the supply voltage. It's a USB charger with 5V - 2.5A output.
As for the SPI speed, I'm not sure how to do that.
What I did is open PxMatrix.h and changed the line like this:
```
  // SPI.setFrequency(20000000);
  SPI.setFrequency(10000000);
```
I also tried changing this part in the .ino patter test
```
  #ifdef ESP8266
    display_ticker.attach(0.001, display_updater); //was 0.002
  #endif
```

Neither case made a difference.

2dom, 2018-06-04:
You appear to collect some noise. Did you connect the node MCU ground with the logic ground of the panel? You could also try wrapping the cables with aluminum foil (careful with shortcuts)

lgaryfallos, 2018-06-04:
Do you mean a GND pin from the MCU to a GND pin of the PI on the panel?

2dom, 2018-06-04:
Yep

2dom, 2018-06-04:
@lgaryfallos Any success?

lgaryfallos, 2018-06-05:
@2dom sadly no.

The current status is:

The latest master (deleted the previous version from ~/Documents/Arduino/libraries,  installed from zip in arduino, which I guess means that the PxMatrix.h that is used is taken from there), changed SPI frequency in PxMatrix.h, then reran the pattern test.

The pattern test is ran with begin(2) and I changed the 0.002 to 0.001 in the attach call.

The only wire I move is the GND to GND which I changed from 1 G pin to another just in case.
When I connect the GND-GND on the MCU, I get other patterns (e.g. in the red filling pattern the rest of the leds are yellow and no movement (or very slow movement)).
Maybe I've done something really wrong with the cabling, but then the original pattern test would not run like it did (I guess) even though it was wrong.

2dom, 2018-06-05:
Sorry ... I am at a loss here. Without having the physical panel to conduct measurements I can't think of a way to sort this out. Something is fishy with the clock/latch singals - pixels appear where they should not.

Do you have a friend with an oscilloscope that you could ask?

lgaryfallos, 2018-06-06:
@2dom I don't, but I'll keep trying, maybe I really did something wrong with the connections.

Thanks for the help though, at the very least someone else with a 1/2 panel can now use this great lib!

closed 2018-06-06 by lgaryfallos