2dom/PxMatrix #13: ADAFRUIT_GFX_EXTRA needs to be implemented better

opticron, 2018-05-23:
https://github.com/2dom/PxMatrix/blob/3ce13fe80681010103fe674c7f53bc4d0e10ead1/PxMatrix.h#L24

This is a handy way to tweak the value during development, but makes it hard to use this library as a user since users don't expect to modify library headers.

One option is to use a static class member with setter function, but this breaks the current usage model since the first place you can modify the value is in setup() which is after construction.

Another option is to add a constructor argument, but the constructor argument list is already getting kind of large. If pruned down, the bare minimum stripped constructor list would be (width, ADAFRUIT_GFX_EXTRA_WIDTH, height). The others could be moved to PxMATRIX.begin() since they're not critical for construction.

2dom, 2018-05-27:
Hi...yeah...I suppose to reshuffle the constructor and begin arguments would be the cleanest approach. However it would break backwards compatibility.

I‘m not that familiar with the include behavior. Is it possible to define ADAFRUIT_GFX_EXTRA_WIDTH in the .ino file and only define it in the header if undefined? That would require a minimal change.

Cheers

Dom

opticron, 2018-06-03:
That would definitely be preferred, but can only be done if this is reworked as a header-only library. This is because if you
`#define ADAFRUIT_GFX_EXTRA_WIDTH 10`
`#include "PxMatrix.h"`
the constant is only defined for the copy of PxMatrix.h included from here. The copy of PxMatrix.h included into PxMatrix.cpp doesn't get that constant unless you somehow define it for the whole project (this can be done with the -D flag if using gcc or g++ directly).

opticron, 2018-06-03:
Looking at the code, the biggest item that should be addressed is making PxMATRIX_buffer a member of the class instead of a global. This also has the benefit of removing global state which is a boon for someone wanting to run multiple panels on a different microcontroller.

opticron, 2018-06-03:
I'm not sure the inline keyword is necessary, but yes to the rest.

2dom, 2018-06-03:
Ok ... done. Check out the current master. The default behaviour is now to set ADAFRUIT_GFX_EXTRA_WIDTH to zero. Would that work?

opticron, 2018-06-04:
Tested and functional. Thanks!

closed 2018-06-04 by opticron