2dom/PxMatrix #14: need some help to get things running

Cyberfly100, 2018-05-26:
Hi,

I am not very familiar with arduino programming, so I hope you excuse my low level questions.
I included the PxMatrix library into a fresh installation of the Arduino IDE via the library manager. I tried running pixeltime.ino on my "P10 (3535) 16x32 - 4S" display hooked up to a amica NodeMCU, but it looks strange.
The 4 in the name made me think the display might be a 1/4 row scan type, so that's how I wired it up according to your instructions. It seems all the leds can light up, but the pattern is somehow mixed up. I attach a video of the pattern_test.ino running. 
I tried to add a line display.setScanPattern(ZIGZAG) after display.begin(4) to test whether it changes anything, but I get a compiling error: 'class PxMATRIX' has no member named 'setScanPattern'.

Any ideas what I might be doing wrong?

Cheers,
Lucas

test pattern: [LINK](https://kth.box.com/v/pixeltime)

2dom, 2018-05-26:
Hi,

are you sure you pulled the latest source from github?

Cheers

Dom

Cyberfly100, 2018-05-26:
I checked the version number of the version that the Arduino IDE pulled for me and it is v.1.0.0.
I also tried to download the latest version from github myself (from the releases tab), but it seems to be the same version and after including library (sketch>>include library>>add .ZIP library) I get the same error if I try to compile the test_pattern file from the folder that was created in the libraries folder (PxMatrix-v1.0.0). Where would I find the setScanPattern function (just so I can check whether my version has it)?

2dom, 2018-05-27:
Hi,

please pull the latest master. The releases do not contain those changes yet.

Cheers

Dom

Cyberfly100, 2018-05-27:
Alright! Now the setScanPattern and setMuxPattern commands work. Thanks for patiently guiding me there :)
After trying around a bit, it seems STRAIGHT and ZAGGIZ give the best result on the pixeltime.ino but it's still not quite working, I think (see pictures).
I thought with a 1/4 row scan display I would not be using the C input. Should I still use the setup line like this: PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C); ?

 I uploaded another video of the test_pattern.ino with 
  display.begin(4);
  display.setScanPattern(ZAGGIZ);
  display.setMuxPattern(STRAIGHT);

testpattern: [LINK](https://kth.box.com/v/pixeltimeSTRAIGHTZAGGIZ)

[image]
[image]
[image]
[image]
[image]
[image]

2dom, 2018-05-27:
You will need A-D...

Cyberfly100, 2018-05-27:
Oh, I get it now and it is also working now :)
In case someone else reads this: I read in the readme that 1/4 row scan only requires A and B input to address the 4 possible rows, but since my display uses STRAIGHT and not BINARY it requires 4 separate pins A, B, C and D to select the current row (which is also mentioned in the readme, but in a paragraph further down). 
Clues for what you should do with your display basically lie in the connector and the name: mine had 4S in the name, which means 1/4 row scan and since it has pins A-D and not only A and B it could have been directly obvious that STRAIGHT should be the correct mode to get it to work. 

Thanks for the help. On to making some fun animations, then.

Cheers!

2dom, 2018-05-27:
Glad that it works ... have fun 😀

irondylis, 2018-05-28:
Hi. I found your project on Hackaday and decided to give it a shot, looks very cool. I bought the LED matrix from the same guys as you on AliExpress, and a NodeMCU board. I'm having a few issues and was hoping you could point me in the right direction. :)

As with Cyberfly's problem, how do you tell if you need A&B or ABCD? My test pattern looks like this:

[https://youtu.be/B0C1g4TqZoQ](url)

I've tried every combo of STRAIGHT/LINE ZIGZAG etc. Currently I'm only using A&B.

Any help appreciated, and thanks for doing the heavy lifting on this. :)

2dom, 2018-05-28:
Thanks...I think Lucas summarized it nicely. Your display ist most likely requires STRAIGHT which necessitates A-D. I will update the readme when I find the time.

2dom, 2018-05-28:
I updated the readme now

closed 2018-05-28 by 2dom

irondylis, 2018-05-28:
Cheer dude.

display.begin(4);
display.setScanPattern(ZAGGIZ);
display.setMuxPattern(STRAIGHT);

worked with the ABCD connections.

Appreciate your help.