2dom/PxMatrix #15: Custom PCB designed for this library

mike-rankin, 2018-05-27:
No issue here but just a quick link to a custom PCB I designed because of this library. Wiring and ESP8266 took a while so I've made a pcb and published the design that is free to all. The git link has a PCB folder that contains the gerber files, schematic and a few other things.

Check it out at: https://github.com/mike-rankin/ESP8266_RGB_Matrix_Cable_Version

https://youtu.be/_pmtngvNnLo

[image]
[image]
[image]

2dom, 2018-05-27:
Hi there,

this is super neat. Much better than the jumper cabling! Great - I will link to it in the readme once I’m back from holiday.

Two ideas for the next Version:
* Reusing the standard flat band connector that comes with the display to connect the two boards (if you separate the signal lines with GND we might even be able to crank up the data rate)
* Move the ESP to the edge of the PCB and let the antenna stand over or make a cutout. This may give some extra range...

Thanks for sharing!

Dom

2dom, 2018-05-28:
ok ... put a link in the readme. Thanks again!

ElectricDesigns, 2018-06-05:
I did another approach to make a PCB compatible with this library where you plug in a Wemos module and use the cables that you'll get with your panels.

https://github.com/ElectricDesigns/ESP8266-Matrix-PCB

2dom, 2018-06-05:
Cool..can you generate PDFs of the schematics and board?

ElectricDesigns, 2018-06-05:
Added some PDF's. Will also add some pictures of the finished boards when I have them made.