2dom/PxMatrix #16: No luck with the 64x64 display

mike-rankin, 2018-06-01:
I have a 64x64 display that has been tested as working on the Adafruit Matrix hat. So far I've had no luck getting it working with this library. It looks like two large bars of black appear no matter what I do. Everything is fine using a 64x32 display but not 64x64.

2dom, 2018-06-01:
Can you post a picture?

mike-rankin, 2018-06-02:
Attached is the 64x64 black screen with purple border. The result is something a bit different. No issues with a 64x32 display connected to the same ESP8266 hardware. This 64x64 has been tested fine on an Adafruit hat.
[image]
[image]

2dom, 2018-06-02:
Can you run the pattern test please ... might be a different layout ...

mike-rankin, 2018-06-02:
Pattern Test and PixelTime.
[image]
[image]

2dom, 2018-06-02:
Hi Mike,

I will need one full cycle (red, yellow,white) of the pattern test as video or gif to see what’s going on...

mike-rankin, 2018-06-02:
I've just recorded a video here: https://youtu.be/z3FURYxryKU

The only changes to examples sketch I've made is uncommenting the bottom 64x64 display and 

`
/uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myWHITE = display.color565(255, 255, 255);

void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.flushDisplay();
  display.setTextColor(myWHITE);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");
`
Changing the display.setTextColor to RED, YELLOW or WHITE does not make a difference since text is never displayed.

2dom, 2018-06-03:
64x64 matrices are usually 1/32 scan. You will have to connect A-E lines and call display.begin(32);

mike-rankin, 2018-06-04:
I can test this today. Do I remove A and E from the ESP8266 pins and just connect them together?

[image]

2dom, 2018-06-04:
Not sure what you mean ... you will have to connect all pins including A,B,C,D,E on the  esp8266 to the panel

2dom, 2018-06-05:
@mike-rankin Ping?

mike-rankin, 2018-06-05:
Right now I have the display wired up as below. It looks like A and E are already connected to GPIO pins. I can modify and test but I'm not sure what to do.

[image]

2dom, 2018-06-05:
OK ... the problem is that the A,B,C,D,E signals are not getting recognized by the panel. As a test ... can you connect A,B,C,D,E to GND and run the pattern test again. This tells the panel to draw something in the 1st and 33rd line.

mike-rankin, 2018-06-06:
I have a custom PCB designed for this display and cannot easily connect or disconnect nets from the display without cutting tracks and then adding jumper wires to connect them again. Right now the 64x32 display works perfectly fine so I'll leave it at that. 

Thanks for all the help along the way.

https://youtu.be/kHkDMiM5ThA

2dom, 2018-06-06:
OK ... I am pretty sure it is those A-F lines if you have the time to investigate ...

closed 2018-06-06 by 2dom

sudheernizwa, 2019-04-13:
is it possible to work 3 units of 64x32 P4 matrix with esp8266 or esp32. i am getting proper display with 2 units but fail with 3 units

themanfrommoon, 2021-11-28:
Hi,
I have the same issue.
How to solve the problem, any suggestions?
https://forum.arduino.cc/t/pxmatrix-mit-64x64-matrix-an-esp32/929993/2

Thank you and best regards,
Chris