2dom/PxMatrix #17: Flickering with 1/8 Scan Panel

jake-b, 2018-06-07:
Hi there.  I'm building a "PixelTimes" clock with a 1/8 scan panel.  I had to modify the initialization to `display.begin(8);` but it works mostly great, but I do get some pretty substantial flickering.

- I tried an ESP8266 at both 80 mhz and 160 mhz.  (It was slightly better at 160mhz) 
- I also tried an ESP32 (which took some minor work to get the PixelTimes project to compile) I believe the ESP32 runs at 240mhz.  The flicker was about the same as the 8266 @ 160 mhz. I have included a video below of the flickering (which you can still clearly make out even with the poor frame rate and color palette of an animated gif)

[image]

I don't suppose you have any suggestions on what might be causing the issue?

Thanks for PxMatrix and your PixelTimes projects. I've enjoyed building one so far.

jake-b, 2018-06-07:
Just as-is.  I haven't tried uncommenting `#define SPIFFS_ENABLE` yet or even getting an openweathermap API key.

Side note, I think power supply is sufficient.  I'm running it off a bench power supply and the voltage is a steady 5v.  When running the animation above draws the most current (of the 3 built in animations) at about 0.9A (on the ESP32)

2dom, 2018-06-07:
Ok good (SPIFFS will result in flicker for sure).

As a test - could you try enabling fast update: set `display.setFastUpdate(true);` after  begin.

Also use display.display(0) in the ticker/ISR function

jake-b, 2018-06-07:
Thanks for the suggestions.  

- On an ESP8266 `display.setFastUpdate(true)` almost fixes the problem at 160mhz.  Just a slight general flicker.  `display.display(0)`fixes the flicker, but seems to reduce the brightness quite a bit.
- On an ESP32 neither fix seemed to do anything and the flicker remained.  I'm not sure why.  I'm wondering if the updates even loaded to the module properly.  I might investigate further later.

I think I can find a happy medium with `display.display` that strikes a nice balance of brightness to flicker.  Maybe a `brightness/40` or something like that.    I think I read 1/8 scan displays are less bright than 1/4 scan displays, so perhaps it is an artifact of the hardware.

Thanks for your help.  Do you have any other thoughts/suggestions?  Otherwise I think we can close this issue. 👍

2dom, 2018-06-08:
You could try reducing the ticker period to 

`display_ticker.attach(0.002, display_updater);`

That can help since it gives the CPU more idle time.

jake-b, 2018-06-08:
Thanks.  Fast update with a 0.002 seems to work nicely. Thanks!

I’m still not sure why the ESP32 still flickers, even with these tweaks.  I’d think that it would be better given the faster clock speed.  I might play with it eventually but for now I’ll continue my build with an ESP8266.

Thanks again for the suggestions.

closed 2018-06-08 by jake-b

2dom, 2018-06-08:
No worries ... have fun with pixeltimes :)

jake-b, 2018-06-09:
Side note for future readers researching ESP32s for PixelTimes.

I tried a few of the things in [this thread](https://github.com/espressif/arduino-esp32/issues/149#issuecomment-275727808).  Specifically:

- adding `#define CONFIG_DISABLE_HAL_LOCKS 1` to `esp32-hal.h`
- Increasing SPI.setFrequency(...) to 40mhz

The first didn't work (flicker remained).  The second seemed to be too fast for the LED panel, and caused odd colors bit the graphics were somewhat recognizable... and the flicker seemed to remain.

"Slow SPI" seems to be a [known thing on the ESP32](https://github.com/adafruit/Adafruit_ILI9341/issues/19) but the usual fixes didn't seem to work for me.  Given the ESP32's dual cores, probably would make sense for a full rewrite fork using an RTOS to take advantage of tasks on both cores.  (Maybe something I'll tackle someday, but probably not)

Again, for now, will just use the tweaks recommended by @2dom on an ESP8266 for my PixelTimes.

Thanks again.