2dom/PxMatrix #18: ESP32: artefacts

sgofferj, 2018-06-09:
I'm working on a little IoT gadget with a P3 64x32 matrix and an ESP32. Everything works fine so far. I do however see artefacts which for lack of better words look like a random line gets displayed every once in a while. It only flashes shortly - I'd say in the millisecond range. I have all the "working" stuff in a function which is pinned to core 0 and my loop() only contains display.display(30), so as far as I understand, it should be the only thing running on core 1.

2dom, 2018-06-11:
Please use the timer function to call display.display (check the PixelTime example). Also, you may want to try display.fastUpdate(true);

sgofferj, 2018-06-12:
Jup, that works, though much less bright. The pixeltime_esp32 example needs some update then because that uses the task method. The normal pixeltime example uses ISR.

2dom, 2018-06-12:
Ok ... thanks for reminding me to update that file. To improve brightness you may also try display.fastUpdate(true) and set display.display(x) where you can adjust x for maximum brightness and minimal flicker. I would start with display.display(10)

sgofferj, 2018-06-12:
Beautiful, thanks! With ISR the param in display() actually has an effect. In the task loop it didn't - not so much at least.

2dom, 2018-06-12:
Great!

closed 2018-06-12 by 2dom