2dom/PxMatrix #19: 32*16 1/2 Scan not correct

Skidouil, 2018-06-11:
Hello , 

I m working with a 32x16 P10 led panel 1/2 scan (this one : https://www.aliexpress.com/item/Freeshipping-outdoor-p10-smd-led-display-module-16x32-32x16-160x320-320x160-1-2-scan-high-brightenss/32669606028.html?spm=a2g0s.9042311.0.0.5e9f4c4dkm11fg )

It seems the 1/2 scan is not working (or not supported ?) pattern test show  the scan like this  (here come beautifull ASCII ART :p )
I use these parameters :
`
display.begin(2);
display.setScanPattern(ZAGGIZ);
display.setMuxPattern(BINARY);

which are the only one to make something similar to test pattern

- = Black;  R = Red ; Y  =Yellow ; W = White

RRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRR
--------------------------------------------
RRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRR
--------------------------------------------
RRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRR
--------------------------------------------
RRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRRR
--------------------------------------------
YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
--------------------------------------------
YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
--------------------------------------------
YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
--------------------------------------------
YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
--------------------------------------------
WWWWWWWWWWWWWWWWWWWWW
--------------------------------------------
WWWWWWWWWWWWWWWWWWWWW

Regards,

2dom, 2018-06-11:
DI you try the current master from github?

Skidouil, 2018-06-11:
I  downloaded the library 2 day ago so i thinked it was the last one but i see  you commited yesterday.
I will try with the new version

2dom, 2018-06-11:
Can you post a video of the pattern test? I will need at least two full (red, yello, white) sequences.

Skidouil, 2018-06-11:
Here is the video you asked for , sorry for the bad quality but i got a crappy phone

https://makeagif.com/gif/led-test-HTbG9i

2dom, 2018-06-12:
Can your check you cabling ... it looks like you set up the chain to be R1G1B1R2G2B2 where it should be R1R2G1G2B1B2

Skidouil, 2018-06-12:
you were right a cable was not secured in the display causing this behaviour.  So the pattern display  is working correctly (exept the display.print("Pixel"); that dont display anything) 

But i will dig into that one.

Skidouil, 2018-06-12:
Btw , thnaks for your time answering my noob question

2dom, 2018-06-12:
No problem … let me know what is going on. There aren’t many people with 1/2 scan displays and I want to make sure that  the library works ok with them.

Skidouil, 2018-06-13:
Here its me again , so here is the feed back : 
-  display.displayTestPattern is working perfectly :  every 2 row got to red => yellow => white
-  displayTestPixel : does work with a pixel moving on the display in red

But when i try to display some text : display.print("Pixel") or display.drawPixel(1,1,myCYAN); dont work

In attach you will find a photo of the display showing "pixel" word.
I got the same result with pixeltime demo the row seems mismatched
[image]

2dom, 2018-06-13:
Try playing with setting the scanpattern (see readme)

Skidouil, 2018-06-13:
Unfortunatly i tried all 6 combination of scanpattern / Muxpattern and everytime the result is wrong

2dom, 2018-06-18:
Ok ... lets try something different. Could you please run the pattern test however substitute

`display.displayTestPattern(30)`

with

`display.displayTestPixel(30)`

2dom, 2018-06-18:
... and send a video of the result :)

Skidouil, 2018-06-20:
Hello , here the file you requested :)

https://makeagif.com/gif/-fQ3dDB

lucianolopesjardim, 2018-06-21:
Hi @Skidouil and @2dom, I'm in the same boat ... I'm trying to control a P10 32x16 1/2 led panel with a esp-wroom-32, it's wired like the diagram but doesn't work.

Here's my test code, follow the link to the video showing the results and the led backpanel/wiring pic:

#include <PxMatrix.h>

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
static int taskCore = 0;

// Pins for LED MATRIX
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);

void setup() {
Serial.begin(9600);
// Define your display layout here, e.g. 1/8 step
display.begin(2);
display.setScanPattern(ZAGGIZ);
display.setMuxPattern(BINARY);
display.fillRect(0,0,8,8,display.color565(0, 255, 0));
}

void loop() {
display.display(30);
//display.displayTestPattern(30);
//display.displayTestPixel(30);
}

Video:
[https://photos.app.goo.gl/1UAqKM6oq5QQuMDs7](url)

Led panel backpanel:
https://photos.app.goo.gl/yxbugHRw6iaMecJ56

 @Skidouil I see you have good progress in your tests, could you post the code of your .ino project?

Skidouil, 2018-06-21:
Hello, 
I use the  pattern_test.ino with configuration adaptation for my panel  , here is the code :
PS i use default scan / mux pattern after trying all combinaison with no success and my panel only use A & B

#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B);
//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(2);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}`

2dom, 2018-06-21:
@lucianolopesjardim New version on master ... let me know what it does

lucianolopesjardim, 2018-06-22:
Thanks guys for your help, I think we're almost there. I connected the esp32 ground to the led panel ground and the flickers just disappeared. I updated the master and did the tests with all the examples, I changed only the following lines:

PxMATRIX display (32,16, P_LAT, P_OE, P_A, P_B, P_C);
display.begin (2);

And here are the results, please take a look:

pattern_test
https://photos.app.goo.gl/Z3EFL7pEmw2jC1Ce6

pixeltime
https://photos.app.goo.gl/m7z8Xdt4yMXg31S99

pixeltime_esp32
https://photos.app.goo.gl/LBojEo51rWKc5HBTA

2dom, 2018-06-22:
OK ... great ... new version on master

Skidouil, 2018-06-22:
Hello,

I update the librairies with the last version and ..... it work perfectly  , thanks you @2dom  for all the work you put in this library.

[image]

I still got a bit of flickering but i suppose it come from my hardware.

2dom, 2018-06-22:
Great! You may want to try

`display.setFastUpdate(true);`

with 

`display.display(20);`

and 160MHz CPU clock.

2dom, 2018-06-23:
@lucianolopesjardim Check https://github.com/2dom/PxMatrix ... currently the library only supports horizontal chaining and only works reliable for panels that use scanning pattern LINE. 

You may want to open a separate issue to support vertical chaining. This will require some time so do though (which I don't have atm)

Closing this issue now ... glad that it finally works

closed 2018-06-23 by 2dom