2dom/PxMatrix #21: Using more than 128x32 pixels

ElectricDesigns, 2018-06-24:
I wanted to use a strain of four 64x32 panels. But the library seems to stop working after I added the third panel to the declaration.

When I declare the width as (128, 32) it works and does everything as it should. Even when I connect the third panel it displays everything over all three of the displays. But if I declare the true size (192,32) it just stops working altogether. Is this a limitation of the library or am I declaring it wrong?

2dom, 2018-06-24:
Yes ... there is a limitation regarding the frame buffer size. It is currently limited to 4096 pixel. You can easily increase that by modifying  

```
#define max_matrix_height 64
#define max_matrix_width 64

```
in the pxmatrix.h file. In your case 

```
#define max_matrix_height 192
#define max_matrix_width 32

```
should work.

ElectricDesigns, 2018-06-24:
I found those statements in the pxmatrix.cpp file. I changed it and will test it in a bit. Thank you!

2dom, 2018-06-24:
... you seem do be on an old version of the library ... you may want to check out the current master

ElectricDesigns, 2018-06-24:
Indeed I was, now it is in the location you specified.

closed 2018-06-30 by 2dom