2dom/PxMatrix #22: using rtc clock on esx12-e

pargit, 2018-06-26:
hi,
i'm trying to add rtc clock but the library uses the i2c protocol (sda/scl uses ports GPIO4/5).
is it possible to move pins A+B to GPIO1/3 ? do i need to change something?

thanks
Mor

2dom, 2018-06-27:
Should work ... just specify in the constructor. You will lose Serial functionality though and will have to comment out Serial.begin / Serial.print.

pargit, 2018-06-28:
Hi.
i'm not that of programmer. can you point how to work with the constructor?
thanks

2dom, 2018-06-30:
Sure ... the constructor is the `PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);` bit. By looking at the the pin definitions

```
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2
```
you just need to change the definition for P_A and P_B to

```
#define P_A 1
#define P_B 3
```

and connect the display accordingly.

CTWF, 2018-08-26:
its work for me : 
`
#define P_LAT D0
#define P_A D1
#define P_B D2 
#define P_C D8
#define P_OE D4

PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B, P_C);

#define SDA D9
#define SCL D3

  Wire.begin(SDA, SCL);
  rtc.begin();
`

closed 2018-10-21 by 2dom