2dom/PxMatrix #25: Brightness setup

pargit, 2018-06-29:
Hi, 
is there a way to control the brightness of the panel?
i searched but couldn't  find any info about it.

thanks
Mor

2dom, 2018-06-30:
Nope ... so far a adjusted brightness via the value x in `display.display(x)`. However, this won't work if you use `display.setFastUpdate(true)`.
I will implement a `display.setBrightness(x)` function when I find the time.

2dom, 2018-07-02:
> Is display.display(x) percent?
>
> If it isn't, What is the min/Max for that?

this is the on-time of the display while scanning in microseconds but only for setFastUpdate(false)

prof955, 2018-07-05:
you are doing Brightness adjust actually while creating test grid:

display.drawLine(xx+16,0,xx+16,5,display.color565(xx*16,0,0));
 
can't we implement this as an setting for all display?

vasquezj-ibm, 2018-07-07:
A brightness function would be great when you get time and thanks for the consideration

2dom, 2018-07-17:
Yes ... brightness is not a biggie ... will do when I have the time

2dom, 2018-08-11:
Ok ... check out the current master. I also updated the example so you can see how it is being used.

closed 2018-08-26 by 2dom