2dom/PxMatrix #26: 64x64 p3 ghosting/double ing

prof955, 2018-07-02:
Hi, 

I am trying to use 64x64 p3 panel.
One piece, single panel.
It's connector starts with R1 pin.
I connected all pins according your explanation(thank to your very detailed connection example)

The problem is panel has ghosting or double ing issue. I don't know how to say in one word but let me explain the issue:

Is I use first 16 line(16x64) everything is ok on this are but next 16 line(16-32x64) is getting same thing i draw/write on first part.
Also it's same for second half of panel too...

It's like I have 4  16x64 panel and first two showing same thing and second two is showing same thing.

If I write "2dom" in first 16x64 I will have:

1* 16x64= -----2dom-----
2* 16x64= -----2dom-----
3* 16x64= -------------------
4* 16x64= -------------------

Did I over-explain? :)

prof955, 2018-07-02:
Also I can't draw/write on second and last part of panel.
They are only copying things from first and third part.

closed 2018-07-04 by prof955

2dom, 2018-07-04:
Just saw this issue on GitHub...did you solver your problem? How?

prof955, 2018-07-05:
**i figured out new function of your library!**

if someone need copy first row to second and third row to fourt they just can use my trick...

#define P_E 2
#define P_OE 0

 :D