2dom/PxMatrix #28: Ghosting Issue

ToXIcGaming, 2018-07-28:
So I have the same panel as [here](https://github.com/2dom/PxMatrix/issues/5) and am using the options detailed there.

The issue im having is ghosting with text and icons, [here](https://imgur.com/a/ww2Heab) are some pics.

ToXIcGaming, 2018-07-29:
Here is the pinout
[image]

and [here](https://gist.github.com/ToXIcGaming/c898f9c670cace0af30cb427d402c4c9) is the code.

ToXIcGaming, 2018-08-03:
@2dom any idea?

2dom, 2018-08-05:
Sorry for the delay ... very busy atm. Can you run the pattern test and post the result (I will need two fully red,yellow,white cycles.

Xonotron, 2018-08-07:
I also have an issue with my panel, this might be related. It's not really ghosting but some columnish crosstalk issue maybe?

I took an overexposed picture to make it clear. To the naked eye, the dim green pixels are only barely visible but there:

[image]

It's also visible in this video, the guy uses PxMatrix but a different panel. So it's propably eighter a bug or a general hardware issue with this kin of panels.?

[image]

2dom, 2018-08-08:
@Xonotron did you try setFastUpdate(true). This would change the timing

ToXIcGaming, 2018-08-11:
Sorry for the late reply, I was away.

[Here](https://i.imgur.com/zUtbd48.gifv) is the result of the pattern test.
I flipped the image because I accidentally had the panel upside down.

2dom, 2018-08-11:
For some reason it enables all 4 rows at a time. Are you running this on ESP32 or ESP8266?

2dom, 2018-08-11:
Looks like a hardware/connectivity issue. To test this disconnect A,B,c,d from the esp and connect A to gnd, b,c,d to vcc. Then re-run the pattern test. You should see a single line.

ToXIcGaming, 2018-08-12:
Doing that produces [this](https://i.imgur.com/kliUwzT.gifv).

2dom, 2018-08-12:
Ok...looking good. Somehow the abcd signals do not reach the matrix...things to try: measure on esp pins with multimeter (a should be low on first pass of pattern test, others high, then b low and so on). Check solder points on esp. try different gpios.

Xonotron, 2018-08-14:
@2dom Yes, this is with setFastUpdate(true).

ToXIcGaming, 2018-09-14:
So I appear to have fixed it but I dont really know how, I ran the code from the thread I originally linked and it worked.

closed 2018-09-29 by 2dom