2dom/PxMatrix #29: Odd panel config

opticron, 2018-08-08:
I just got another set of 1/8 scan 32x32 panels from China and the're even weirder than the first set I got. I've poked at the lib trying to find settings that will make it work right, but I figured I'd ask. Do we have settings that can handle this kind of pattern: https://photos.app.goo.gl/kzqTATvhVoxiHZGy9 ?

shodan45, 2018-08-09:
I **think** my 32x32 panel is doing the same:

https://photos.app.goo.gl/gFottGJd43rvuzQA8

I bought mine from Sparkfun, if that helps.

opticron, 2018-08-10:
Yours is doing something different from mine, but still weird and probably currently unsupported.

opticron, 2018-08-12:
@shodan45 Alright, I have some code to test tomorrow which should contain the solutions for both our panel setup issues. I'll let you know once I've verified it works for mine.

opticron, 2018-08-12:
@shodan45 when you get a chance, could you test against my fork with setScanPattern(VZAG) ? https://github.com/opticron/PxMatrix

opticron, 2018-08-13:
@shodan45 I just pushed another commit, pull it and give it a try. Happy to help, these height/pattern=4 panels are getting pretty damn weird.

shodan45, 2018-08-13:
That almost did it. I thought all that was needed was to fix the "y chunks" problem, but the X axis isn't right either. It seems to be mapping to 16-31, then 0-15.

`      display.drawPixel(0, 0, myRED);
      display.drawPixel(16, 0, myYELLOW);
      display.drawPixel(31, 0, myGREEN);
` gives:
[image]

I stuck a quick hack of `x -= 16; if (x < 0) x += 32;` in there, and that fixed it, but I'm sure that's not the best way to fix it.

opticron, 2018-08-13:
Alright, I'll update my branch and the PR today with the right values.

shodan45, 2018-08-13:
Odd, that didn't seem to do anything... I double checked that I have the latest version, and I do. It's still doing the 16-31, 0-15 thing on the X axis.

opticron, 2018-08-13:
Alright, well I just inverted the X blocks again on my branch just for good measure. Hopefully that gets it done. Last time I simplified some code in the process and may have ended up inverting things twice.

shodan45, 2018-08-14:
That seems to have fixed it! Thank you!!

closed 2018-08-15 by opticron