2dom/PxMatrix #32: Multi Panel > 3 x P10(16x32)

CTWF, 2018-08-26:
I use RGB matrix led panel from China with specifications:
- Port: hub 75
- Scan: 1/8
Hardware : Nodemcu ESP8266 Lolin
The result is normal for multi panel = 2. the rest (panel_width> 2) the result is the text / pixel shifts approximately 2 pixels for each color.

**the code that I use:**

`
#include <PxMatrix.h>
#include <Ticker.h>

#define P_LAT D0  //D0
#define P_A D1     //D1
#define P_B D2     //D2
#define P_C D8    //D8
#define P_OE D4    //D4

#define PxMATRIX_MAX_WIDTH 96
#define PxMATRIX_MAX_HEIGHT 16
#define panel_width  3

PxMATRIX display(96, 16, P_LAT, P_OE, P_A, P_B, P_C);

//#define double_buffer 0

uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

uint16_t myCOLORS[8] = {myRED, myGREEN, myBLUE, myWHITE, myYELLOW, myCYAN, myMAGENTA, myBLACK  }; //{myRED,myGREEN,myBLUE,myWHITE,myYELLOW,myCYAN,myMAGENTA,myBLACK};
Ticker display_ticker;

void display_updater() {
  display.display(70);
}

void setup() {
  Serial.begin(9600);
  display.begin(8);
  display.setPanelsWidth(panel_width);
  display_ticker.attach(0.002, display_updater);
  display.setFastUpdate(true);
  display.clearDisplay();
  display.setTextWrap(false);
  display.fillScreen(myBLACK);
}
void loop() {
  for (byte x = 0; x < 8; ++x) {
    display.setTextColor(myCOLORS[x], myBLACK);
    display.setFont();
    display.setCursor(0, 0);
    display.setTextSize(1);
    display.print("3 x P10 RGB");
    delay(1000);
  }
  yield();
}
`
image :  
[image]

2dom, 2018-08-26:
@opticron Do you have time to look into this? Appears to be a minor offset calculation problem ... also see https://twitter.com/chellotech22/status/1033117086283784193

opticron, 2018-08-26:
I'll take a look. @CTWF the title of the issue says 16x32, I assume these are actually 32x16 (WxH)?

opticron, 2018-08-26:
It's almost like the color offset is wrong. @CTWF could you provide a pattern test video to verify it's just the mapping that's bad?

CTWF, 2018-08-27:
@opticron  yes 32x16 (WxH)
test pattern look like everything is OK.
image : [image]

found other issue : 
when setfont with height > 8 pixel OR draw vertical line from y = 0 to y = 15
image : [image]

CTWF, 2018-08-29:
@opticron how to update offset color with parameter :
g = r+2
b = r+4 or g+2

opticron, 2018-08-29:
That's pretty odd. A bad initial mapping can't cause color offsets like that, it would just cause the pixels to show up in the wrong places but the colors would still map to the same location. The code for calculating the color offset hasn't changed since first commit and I doubt it's wrong.

It's almost as if one of your panels has an extra hidden or a missing column or the PxMatrix library was initialized with the wrong size. Additionally, there looks to be some weirdness going on between the first and second panels in your initial picture when you're showing only blue or only green text.

I know it's probably a lot to ask, but could you verify those two panels individually? Maybe the end of the chain as well.

CTWF, 2018-08-31:
Still stuck with panel brand Qiangli

lucianolopesjardim, 2018-10-05:
Hi @CTWF , how are you? I'm having the same problems as you, I'm trying to use 4 and 8 chained panels brand lightall... Did you have any progress changing the library?

ebelouet, 2018-12-21:
HI
I have same mistake, 2 panels P10is perfect, with 3 panels I have mistake.
Anybody find a solution for this ?

Thank you

2dom, 2018-12-21:
Are these all the same issue?

@metingul @ebelouet try setting the correct matrix size by placing

```
PxMATRIX_MAX_WIDTH 128
PxMATRIX_MAX_HEIGHT 32
```

before `#include <PxMatrix.h>` in your sketch

ebelouet, 2018-12-21:
thank for answering but no mistake change, here is code and pictures :

#include <Ticker.h>

#include <PxMatrix.h>
// The library for controlling the LED Matrix
// Needs to be manually downloaded and installed
// https://github.com/2dom/PxMatrix

Ticker display_ticker;

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

// PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
// PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
PxMATRIX display(96, 32, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E);
#define PxMATRIX_MAX_WIDTH 96
#define PxMATRIX_MAX_HEIGHT 32
#define panel_width 3
uint16_t myRED = display.color565(255, 0, 0);
// This is a horribly inefficient way of drawing this, but meh.. :)
// Converted using the following site: http://www.rinkydinkelectronics.com/t_imageconverter565.php
uint16_t static banana_frame_1[] = {
0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, // 0x0010 (16) pixels
0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, // 0x0020 (32) pixels
0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, // 0x0030 (48) pixels
0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, // 0x0040 (64) pixels
0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, // 0x0050 (80) pixels
0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, 0xF800, // 0x0060 (96) pixels
0xF800, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0070 (112) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0080 (128) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0090 (144) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00A0 (160) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00B0 (176) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xF800, // 0x00C0 (192) pixels
0xF800, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00D0 (208) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00E0 (224) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00F0 (240) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0100 (256) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0110 (272) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xF800, // 0x0120 (288) pixels
0xF800, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0130 (304) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0140 (320) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0150 (336) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0160 (352) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0170 (368) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xF800, // 0x0180 (384) pixels
0xF800, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0190 (400) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01A0 (416) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01B0 (432) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01C0 (448) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01D0 (464) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xF800, // 0x01E0 (480) pixels
[... cut ...]
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0270 (624) pixels
[... 294 lines cut ...]
display.clearDisplay();

display_ticker.attach(0.002, display_updater);
yield(); 
}

void drawFrame(uint16_t *frame) {
display.clearDisplay();
int imageHeight = 32;
int imageWidth =96;
int counter = 0;
for (int yy = 0; yy < imageHeight; yy++) {
for (int xx = 0; xx < imageWidth; xx++) {
display.drawPixel(xx, yy, frame[counter]);
counter++;
}
}
delay(200);
}

void loop() {
drawFrame(banana_frame_1);
//display.drawPixel(0,0,0xFFFF);
yield();
}

ebelouet, 2018-12-21:
in fact, we have carre but for 2 panels but with 3 panels we have mare pixels.

2dom, 2018-12-21:
Please re-read my previous comment...

ebelouet, 2018-12-21:
yes i also put 
PxMATRIX_MAX_WIDTH 128
PxMATRIX_MAX_HEIGHT 32

before but no change.

2dom, 2018-12-21:
You put them in the wrong place ...

ebelouet, 2018-12-21:
here is your request but nothing change.

#define PxMATRIX_MAX_WIDTH 96
#define PxMATRIX_MAX_HEIGHT 32

#include <PxMatrix.h>
#include <Ticker.h>
// The library for controlling the LED Matrix
// Needs to be manually downloaded and installed
// https://github.com/2dom/PxMatrix

Ticker display_ticker;

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

// PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
// PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
PxMATRIX display(96, 32, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E);

#define panel_width 3
uint16_t myRED = display.color565(255, 0, 0);
// This is a horribly inefficient way of drawing this, but meh.. :)
// Converted using the following site: http://www.rinkydinkelectronics.com/t_imageconverter565.php
uint16_t static banana_frame_1[] = {
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, // 0x0010 (16) pixels
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, // 0x0020 (32) pixels
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, // 0x0030 (48) pixels
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, // 0x0040 (64) pixels
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, // 0x0050 (80) pixels
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, // 0x0060 (96) pixels
0xFFFF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0070 (112) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0080 (128) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0090 (144) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00A0 (160) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00B0 (176) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xFFFF, // 0x00C0 (192) pixels
0xFFFF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00D0 (208) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00E0 (224) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x00F0 (240) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0100 (256) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0110 (272) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xFFFF, // 0x0120 (288) pixels
0xFFFF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0130 (304) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0140 (320) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0150 (336) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0160 (352) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0170 (368) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xFFFF, // 0x0180 (384) pixels
0xFFFF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0190 (400) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01A0 (416) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01B0 (432) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01C0 (448) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x01D0 (464) pixels
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0xFFFF, // 0x01E0 (480) pixels
[... cut ...]
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, // 0x0260 (608) pixels
[... 164 lines cut ...]
display.clearDisplay();

display_ticker.attach(0.002, display_updater);
yield(); 
}

void drawFrame(uint16_t *frame) {
display.clearDisplay();
int imageHeight = 32;
int imageWidth =96;
int counter = 0;
for (int yy = 0; yy < imageHeight; yy++) {
for (int xx = 0; xx < imageWidth; xx++) {
display.drawPixel(xx, yy, frame[counter]);
counter++;
}
}
delay(200);
}

void loop() {
drawFrame(banana_frame_1);
//display.drawPixel(0,0,0xFFFF);
yield();
}

2dom, 2018-12-21:
What size are your panels?

ebelouet, 2018-12-21:
nothing change

2dom, 2018-12-21:
Are you using the latest GitHub master ?

ebelouet, 2018-12-21:
1.3.0 on arduino

2dom, 2018-12-21:
Ok...can you check that there are no cpp files in the library folder. Otherwise everything looks goid

2dom, 2018-12-21:
Can you try drawing a single pixel instead of the banana

ebelouet, 2018-12-21:
yes, 1 red at 0.0, I have it, but inside i have in col 7 red in midille bright.
with 1 white at 0.0, I have it but after in line I have 4 leds in blue en 7 led in colone 2 in green.

2dom, 2018-12-21:
Can you post a picture?

ebelouet, 2018-12-21:
on can't on this web page, do you have email for that ?

ebelouet, 2018-12-21:
i hev sending picture by email.

lucianolopesjardim, 2018-12-22:
Hey guys other people like me have the same issue... please share pictures and information! Tks!

ebelouet, 2018-12-22:
yes but i can't put picture on this web site...

CTWF, 2019-01-04:
In my case. My panel is uniqe and discontinue from factory. 3 panels is normal. Issue appear on 4 panels.

lucianolopesjardim, 2019-01-04:
@CTWF @ebelouet My panels are p10 rgb 32x16 1/8 Chinese brand LIGHTALL and the problem happens with 4 or more panels .... with 3 panels works perfectly.

lucianolopesjardim, 2019-01-04:
Hey @2dom @opticron I have 16 panels p10 rgb 32x16 1/8 Chinese brand LIGHTALL and I'm available to do the necessary tests to solve this problem. I'm using the esp-wroom-32. Waiting for instructions! Thanks.

ebelouet, 2019-01-04:
[white_rectangle9632.zip](https://github.com/2dom/PxMatrix/files/2727855/white_rectangle9632.zip)
[image]
here is code source and picture for 3 32x32 panels
I am drawing single pixel line ouside and smal in middle but we have another pixel blue.

Sincerely
eric

2dom, 2019-01-04:
@lucianolopesjardim At some point you will have to use another library - PxMatrix is not made for chaining a a large number of panels. I advise to use something that drives all RGB lines in parallel, preferably using DMA.

ebelouet, 2019-01-04:
Thank, do you have link example for that ?

Eric

2dom, 2019-01-04:
Never tried that one but sounds promising: https://github.com/pixelmatix/SmartMatrix/tree/teensylc

lucianolopesjardim, 2019-01-04:
@2dom I'll take a look, thanks for the suggestion.

CTWF, 2019-01-05:
I found other issue when i use panel 64x32 RGB more than 2 panel (exception error).

But when running 2 panel, everything is normal

ebelouet, 2019-01-05:
I think it would be nice to give the max configuration of panels for users

So, could you explain where is limitation, speed or 3v3 signal for lenght ?

m000lie, 2025-11-15:
hi, im having trouble understanding how you managed to chain 3 displays while using the serial input method. do you mind posting an image of how you wired them?

[image]