2dom/PxMatrix #33: Is there a way to change font size?

ToXIcGaming, 2018-09-15:


ToXIcGaming, 2018-09-15:
So I have figured out its possible to do it using adafruit GFX.

CTWF, 2018-09-15:
I make custom font with Adafruit font creator

ToXIcGaming, 2018-09-15:
So I can use the fonts but when using a certain font (TomThumb) there is a problem, as you can see [image]

when using a different font (Picopixel), it works fine [image]

I have no idea why this is happening.

CTWF, 2018-09-15:
Not yet try this font. Because, i made custom font to fit my need.

2dom, 2018-09-15:
There is an issue with the display width that is passed to the AdafruitGFX library - if it is set correctly TomThumb will wrap prematurely.

You can fix this by adding

`#define ADAFRUIT_GFX_EXTRA 10`

before

`#include <PxMatrix.h>`

closed 2018-09-16 by ToXIcGaming