2dom/PxMatrix #34: Scroll text / clear part of display

ptee132, 2018-09-17:
Hey guys, 

first of all: nice work. The library works very well. I'm totally happy how easy it is to set up. 
I have 2 questions - no issues per se:

1. Is it possible to scroll text with a command like display.scroll("Text");? If not, how could I realize it?
2. Is it possible to clear parts of the display. For example the upper half or another specific part of the display. For data updates I always have to clear the whole display and draw everything again. Is there a better solution?

Thanks for your help. 

Best 
Patrick

2dom, 2018-09-17:
1. No ... the library does not implement scroll. You could achieve this by drawing text at position x, delay, clear display, drawing text at position x+1, delay, ...
2. Depends what you mean with 'clear'. If you assume that the clear state of the display is all LEDs of then you could just draw a black rectangle over the areas you want cleared...

ptee132, 2018-09-17:
Thanks for the quick reply. 
Good idea to use black rectangles. Could you tell me how to draw them?

2dom, 2018-09-17:
For example check out https://github.com/adafruit/Adafruit_ILI9341/blob/master/examples/graphicstest/graphicstest.ino how or https://cdn-learn.adafruit.com/downloads/pdf/adafruit-gfx-graphics-library.pdf on how to use AdafruitGFX

closed 2018-09-17 by ptee132