2dom/PxMatrix #35: Another 1/4 scan issue

laplacier, 2018-09-20:
I purchased a P10 1/4 scan matrix with the straight multiplexing pattern (ABCD) and I'm having issues getting the pattern_test code to display properly. Colors are correct, # of columns are correct, but rows are not correct. I've tried all combinations of LINE/ZIGZAG/ZIGGAZ and BINARY/STRAIGHT without success. I've triple checked the wiring (R1 top left corner) and everything tests okay there. 

Video:
https://gfycat.com/InnocentBitterCornsnake

I'm using an ESP32 microcontroller: LOLIN D32 Lite

2dom, 2018-09-20:
Hm ... almost looks like the address signals (ABCD) are active high instead of active low.
Could you try placing all "value==" with "value!=" in the following section of PxMatrix.h (PxMATRIX::set_mux(uint8_t value)

```
if (_mux_pattern==STRAIGHT)
  {
    if (value==0)
      digitalWrite(_A_PIN,LOW);
    else
      digitalWrite(_A_PIN,HIGH);

    if (value==1)
      digitalWrite(_B_PIN,LOW);
    else
      digitalWrite(_B_PIN,HIGH);

    if (value==2)
      digitalWrite(_C_PIN,LOW);
    else
      digitalWrite(_C_PIN,HIGH);

    if (value==3)
      digitalWrite(_D_PIN,LOW);
    else
      digitalWrite(_D_PIN,HIGH);
  }
```

laplacier, 2018-09-20:
Alright, new results

https://gfycat.com/JauntyTalkativeCuttlefish

1st: Row 0/4 not lit
2nd: Row 1/5 not lit
3rd/4th/5th: All rows lit

BVVLab, 2018-10-25:
Hello! I purchased same P10 panel. 
 I've tried all combinations of last lib PxMatrix enum scan_patterns {LINE, ZIGZAG, ZAGGIZ, WZAGZIG, VZAG}; without success.
Test with LINE and printing P
  display.setScanPattern(LINE);
  display.setMuxPattern(STRAIGHT);  
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("P");
Result photo. https://www.dropbox.com/s/wao37n4fo6inpro/P10_LINE_printP.jpg?dl=0
Wiring is ok, because with the previous panel works great.
It's panel photo https://www.dropbox.com/s/k81ie6enxd2p42p/P10Prev.jpg?dl=0

ALEX-LED, 2018-10-27:
>Wiring is ok, because with the previous panel works great.

analogically! 

hi friends!
thanks a lot to the author of the project for the work on the library!

bought from one seller such modules, three types, 1\2P 10, 1\4 P 10, and 1\5 P 8-scan does not work
[URL=http://vfl.ru/fotos/d5b5577b23957074.html][IMG]http://images.vfl.ru/ii/1540625787/d5b5577b/23957074_m.jpg[/IMG][/URL]

[URL=http://vfl.ru/fotos/519ce37e23957081.html][IMG]http://images.vfl.ru/ii/1540625846/519ce37e/23957081_m.jpg[/IMG][/URL]

prior to that, used the other 1\4 R 10 Hoozoe 
[URL=http://vfl.ru/fotos/34fa7bc723957115.html][IMG]http://images.vfl.ru/ii/1540626206/34fa7bc7/23957115_m.jpg[/IMG][/URL]

scan ZAGGIZ - work perfectly
[URL=http://vfl.ru/fotos/25775afb23957114.html][IMG]http://images.vfl.ru/ii/1540626162/25775afb/23957114_m.jpg[/IMG][/URL]

I understand that the case in the picture of the scan, but I don't know how to make your own version of the scan

I can draw a picture - I can even compare these modules using a special controller-maybe together we can expand this list - / / LINE, ZIGZAG, ZAGGIZ WZAGZIG VZAG

2dom, 2018-10-27:
Hi ... could you run the pattern test with

`display.displayTestPixel(70);`

and post the video?I will need a least one full red,yellow,white cycle ....

ALEX-LED, 2018-10-27:
Yes, of course, the video link - https://www.youtube.com/watch?v=849gKAEk3MQ&feature=youtu.be

Is that enough?

thank you for your quick response to the request

I have not yet been able to draw a picture of this module

I thought it would help.

(it's a drawing module, which works well - hoozoe)
http://vfl.ru/fotos/f8aacec923966731.html

2dom, 2018-10-27:
Wow ... this is very different to all other panels that I have seen so far. Probably not gonna be able to support that one anytime soon I am afraid.

Where did you get the panel from?

ALEX-LED, 2018-10-28:
here's another matrix, this time 1\2 scan, please see the video https://www.youtube.com/watch?v=3-KpHG0qQL4

I think he is better than what I previously showed, maybe it can work
do you think?

BVVLab, 2018-10-31:
My solution for this matrix https://imgur.com/a/rW4C95N is:
1. Correction In file  PxMatrix.h:
inline void PxMATRIX::Transformation(int16_t x,int16_t y,int16_t* new_x,int16_t* new_y) {
	*new_x=x;*new_y=y;

	if((y>=0)&&(y<4)|(y>=8)&&(y<12))
	{
		if(x<4){*new_x=x;*new_y=y+4;}
		if((x>=4)&&(x<8)){*new_x=x+8;*new_y=y+4;}
		if((x>=8)&&(x<12)){*new_x=x+8;*new_y=y+4;}
		if((x>=12)&&(x<16)){*new_x=x+16;*new_y=y+4;}
		if((x>=16)&&(x<20)){*new_x=x-16;*new_y=y;}
		if((x>=20)&&(x<24)){*new_x=x-8;*new_y=y;}
		if((x>=24)&&(x<28)){*new_x=x-8;*new_y=y;}
		if((x>=28)&&(x<32)){*new_x=x;*new_y=y;}		
	}
	if((y>=4)&&(y<8)|(y>=12)&&(y<16))
	{
		if(x<4){*new_x=x+4;*new_y=y;}	
		if((x>=4)&&(x<8)){*new_x=x+4;*new_y=y;}	
		if((x>=8)&&(x<12)){*new_x=x+12;*new_y=y;}
		if((x>=12)&&(x<16)){*new_x=x+12;*new_y=y;}
		if((x>=16)&&(x<20)){*new_x=x-12;*new_y=y-4;}
		if((x>=20)&&(x<24)){*new_x=x-12;*new_y=y-4;}	
		if((x>=24)&&(x<28)){*new_x=x-4;*new_y=y-4;}
		if((x>=28)&&(x<32)){*new_x=x-4;*new_y=y-4;}		
	}	
}

inline void PxMATRIX::drawPixelRGB565(int16_t x, int16_t y, uint16_t color, bool selected_buffer) {
  uint8_t r = ((((color >> 11) & 0x1F) * 527) + 23) >> 6;
  uint8_t g = ((((color >> 5) & 0x3F) * 259) + 33) >> 6;
  uint8_t b = (((color & 0x1F) * 527) + 23) >> 6;
  int16_t new_x,new_y;
  Transformation(x,y,&new_x,&new_y);
  x=new_x;y=new_y;
  fillMatrixBuffer( x,  y, r, g,b, selected_buffer);
}

inline void PxMATRIX::drawPixelRGB565(int16_t x, int16_t y, uint16_t color) {
  uint8_t r = ((((color >> 11) & 0x1F) * 527) + 23) >> 6;
  uint8_t g = ((((color >> 5) & 0x3F) * 259) + 33) >> 6;
  uint8_t b = (((color & 0x1F) * 527) + 23) >> 6;
  int16_t new_x,new_y;
  Transformation(x,y,&new_x,&new_y);
  x=new_x;y=new_y;  
  fillMatrixBuffer( x,  y, r, g,b, 0);
}
  inline void Transformation(int16_t x,int16_t y,int16_t* new_x,int16_t* new_y);

2. In sketch file:
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);
display.setScanPattern(LINE);
display.setMuxPattern(STRAIGHT);

2dom, 2018-10-31:
Great that you got it working! If you would like to see it as part of the library it has to look a b it differnt though though. You would need to introduce a new line scanning pattern, e.g. VLINE, and augment the offset calculation code.

ALEX-LED, 2018-10-31:
**BVVLab** thank you for your experience, I will study it

closed 2019-02-22 by 2dom