2dom/PxMatrix #36: Flickering and double buffer ?

pyzimmer, 2018-09-21:
Hello, i need to refresh a digit every 100 ms. But the whole display is flickering. why ?
see video : https://photos.app.goo.gl/GgTtfJbY33aCcLqN8
Is that an issue with double buffering ? How to use double buffer ?
Perhaps issue with the only one color (white leds) display -> how to refresh only 1 color ?
I use :
- Wemos D1 ESP 8266 (160 Mhz)
- one color (white) display, leds are white only.

my code :
#include <PxMatrix.h>

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#define double_buffer true

// Pins for LED MATRIX
PxMATRIX display(96,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);

// ISR for display refresh
void display_updater()
{
  display.display(70); // temps d'affichage en microsecondes
}

uint16_t BLANC = display.color565(0, 0, 0);
uint16_t NOIR = display.color565(255, 255, 255);

void setup() {
  display.begin(16); //{4,8,16,32}
  //display.setFastUpdate(true);
  display_ticker.attach(0.002, display_updater);  // fréquence de rafraichissement en ms
  display.setTextColor(BLANC, NOIR);
  display.setTextSize(2);
  //display.setTextWrap(false); // Allow text to run off right edge
  //display.flushDisplay();
  display.fillScreen(NOIR);
}

void loop() {
  display.setCursor(0,0);
  //display.fillScreen(NOIR);
  display.print("12345678");
  delay(100);
  display.setCursor(0,0);
  //display.fillScreen(NOIR);
  display.print("12345608");
  delay(100);
}

thank you

CTWF, 2018-10-09:
Try
display.setTextColor(NOIR,BLANC) before print text

pyzimmer, 2018-10-11:
Thank you,
Colors are inverted, but  flickering too. 
How to use double buffering ? is set #define double_buffer true , but no change...

2dom, 2018-10-12:
double_buffering is a bit experimental atm but you would have to write someting linke this:

```
void loop() {
    display.setCursor(0,0);
    display.print("12345678");
    display.selectBuffer(1); // Display from buffer 0, draw to buffer 1
    delay(100);
   display.setCursor(0,0);
   display.print("12345608");
    display.selectBuffer(0); // Display from buffer 1, draw to buffer 0
    delay(100);
}
```

pyzimmer, 2018-10-16:
Thank you, but your code flicker too, sorry

2dom, 2018-10-21:
Sorry ... your will have to enable double buffering first:

add 

`#define double_buffer`

before

`#include <PxMatrix.h>`

olesk75, 2019-02-22:
I'm dusting off old skills after nearly a decade away from C, so forgive me if what I'm asking here is trivial, but I'm looking for a faster way of copying chunks of data to the buffer. 

I'm essentially playing with eight sprites to make a moving animation. Would it be as simple as using memcpy/memcpy_P, or does one have to convert the data into a format the display can understand *before* copying it to the buffer?

Or put more simply: if you wanted to repeatedly copy a block from memory to different locations on the display, what would be the most optimal way of going about it?

2dom, 2019-02-22:
Its a good question. The short answer is no ... there is no efficient way of copying data into the frame buffer, e.g. with memcpy. The long answer is as follows: the data that is being sent to the display is rather complex and not just a serialized version of the image you see. This gives you two choices on when you compute the (complex) position of every pixel in the serial data stream sent to the display - either on drawing a pixel to the frame buffer or on updating the display from the frame buffer.

I opted for computing pixel positions at draw since then the frame buffer is in a format that allows me to update the display very quickly (just pumping out the entire buffer via SPI bus). The disadvantage is that drawing to the frame buffer is slow however this is less important since display updates happen much more frequently than buffer draws. 

To speed things up a bit I implemented double buffering that allows you to draw to the frame buffer in the background while the display is updating from a second buffer. This should give you sufficient performance for non-flickering animations. Having said this - I have not actually used it yet. So let me know if it works for you.

olesk75, 2019-02-22:
Thanks Dominic - very helpful. I'll play around with the double buffering a bit and see how it goes. 

Just to be 100% clear I'm getting this right though - if I understand you correctly there is no faster way to get for example a 20x32 color sprite from memory to the display than to loop over drawPixelRGB565() 20x32=640 times?

2dom, 2019-02-22:
Correct!

olesk75, 2019-02-22:
Understood - thanks!

olesk75, 2019-02-22:
My conclusion so far is that for sprites at least (see below) double buffering makes no difference to flickering at all, which is fine as it's not really a problem in this case. The animation artifacts are due to the mobile phone camera - in person Mario is completely steady and the colors correct and constant. I'm guessing it's probably a bigger issue for drawing text since that's probably slower and spans across several display refresh cycles. In my case I'd rather save the pretty big chunk of memory double buffering consumes. Anyway, thought I should mention it.
[image]

2dom, 2019-02-23:
Cool...thanks for the info!

closed 2019-02-23 by 2dom