2dom/PxMatrix #40: Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)

mrfaptastic, 2018-10-10:
Hi.

This awesome library isn't safe to use on the ESP32 when using a hardware interrupt timer to refresh as any form of interrupt-generating WiFi or Flash-Read/Write access will cause a Core 1 panic. This is due to the fact the main PxMatrix::display() function is not in the ESP32's IRAM (interrupt function memory).

Having the usual display_update() function in IRAM_ATTR is not sufficient. However, changing the library such that 

```
void PxMATRIX::display(uint16_t show_time) {

becomes something like

void IRAM_ATTR  PxMATRIX::display(uint16_t show_time) {

```

... results in it failing to compile, as I believe the whole class will need to be refactored to insure this function is a static singleton.

My sketch used a combination of WiFiManager and Webserver, and for example, if I ever accessed by ESP32 via HTTP (which results in it reading a file of the SPIFFS Flash or even PROGMEM-based hard-coded string), there would be an immediate crash.

SDK seems to imply any ISR code must be stored in the IRAM using via the IRAM_ATTR flag.
Ref: https://dl.espressif.com/doc/esp-idf/latest/api-reference/system/intr_alloc.html

Sanman96, 2018-11-19:
Bump. Also having this issue. Any updates on this? The other library mentioned above has flickering issues and I would prefer to use this library instead.

mrfaptastic, 2018-11-19:
Well, if you prefer to try yet another library (though different wiring will be required). I wrote one to leverage the DMA capabilities of the ESP32, though only tested with 64x32 panels.

https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA

2dom, 2018-11-20:
I don't have time atm to fix this ... do you guys wanna give it a shot?

closed 2019-02-22 by 2dom

metingul, 2019-03-10:
Can we stop to PXMatrix display? I had having problem with DMD2 for single color when access to EEPROM or SPIFFS , I solved that problem with dmd.end() function. and after write eeprom or spiffs I start to again DMD. 
I tried to display.end(); function bu no function available.

2dom, 2019-03-10:
Simply stop the display refresh loop...