2dom/PxMatrix #42: Examples

ToXIc-Dev, 2018-10-19:
Would be useful if some more examples could be added, similar to the ones on the [Raspberry PI](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use) version.

2dom, 2018-10-21:
... no time for more examples atm :( ... wanna create a few ? :)

2dom, 2018-10-22:
Haha...come on...I believe in you 😀

tablatronix, 2019-01-17:
I have ported all adafruit examples, I will try to PR them this week

2dom, 2019-02-22:
Did you in the end?

tablatronix, 2019-02-22:
Sorry my matrix was put away for a bit, I will revisit soon, I can post what I have to gist for now

justsilviu, 2019-10-09:
Hello.
A simple picture with how the wiring for multiple panels in the readme would help.
i understand how  2 or 3 in a single row work, and i have tried and it works.
but how about creating a matrix of 3x3, how should that be wired?
and "display.setPanelsWidth();" function does not have a corespoding display.setPanelsHeight(); ??
thanks

ToXIc-Dev, 2019-10-09:
> Hello.
> A simple picture with how the wiring for multiple panels in the readme would help.
> i understand how 2 or 3 in a single row work, and i have tried and it works.
> but how about creating a matrix [...]

This is not really related to this issue, you should create a new issue.

ToXIc-Dev, 2019-10-09:
> Sorry my matrix was put away for a bit, I will revisit soon, I can post what I have to gist for now

Would be great if you could.

2dom, 2019-10-19:
Any news on those examples - did I miss a pull request?

tablatronix, 2019-10-19:
I just found them yesterday, I never commited it to my fork. I will clean them up and PR it.

2dom, 2019-10-19:
Great

closed 2020-03-28 by 2dom

tablatronix, 2020-03-28:
haha geez what a year
I totally kept dropping the ball on this, I actually JUST got around to touch back to this project I was using this for..