2dom/PxMatrix #43: TextAlignment GFX not working in PXmatrix library

RobertHolscher, 2018-10-21:
Dear All,
I am trying to use the TextAlignment function of GFX in the display PxMatrix library but this GFX function is not available. Is there an alternative present in the library to use to get a center aligment on the text instead of a left top corner alignment?

2dom, 2018-10-21:
Hi ... can you point to where this is implemented? Cannot find anything in AdafruitGFX.

RobertHolscher, 2018-10-21:
2dom
Here is a reference I have found on the internet
https://forum.pjrc.com/threads/31939-ILI9341_t3-draw-centered-text-string

2dom, 2018-10-21:
Doesn't seem to be part of the standard AdafruitGFX - you could try adding that section to your local copy of AdafruitGFX. This issue is not related to PxMatrix....closing

closed 2018-10-21 by 2dom