2dom/PxMatrix #45: GetPixel

Aircoookie, 2018-10-30:
Awesome job on the library! Thank you!

I've noticed this function isn't implemented:
```
// Does nothing for now
  uint8_t getPixel(int8_t x, int8_t y);
```

What are the challenges/hurdles if one was to properly implement the function? In my application I'd like to shift pixels around (for example set a pixel to the color of its left neighbor). For that, getPixel() is essential if I don't want to use an auxiliary buffer array that would cost at least 4kb additional RAM.
I'll understand should there are obvious reasons why this isn't (easily) possible, just asking.

2dom, 2018-10-31:
The problem is that the data structure that keeps the images is essentials crafted so it can be sent to the matrix ‚as is‘, I.e. requiring no manipulation during an update which reduces latency / flicker.

The down-side of this is that individual pixel information is spread over multiple locations in the frame buffer which would require individual offset calculations etc to retrieve.

Long story short...it is possible to implement this albeit the easier way and faster way would be to keep the pixel data in a separate array

Aircoookie, 2018-11-01:
Thanks for your explanation! Yeah, i figured it'd be something like that. The function would also probably need to take things like colordepth and scan pattern into account if the data structure contains the raw bits pushed out to the panel.

2dom, 2018-11-01:
Yeah ... it's a lot of effort for saving a few kB of RAM ...

I will keep the issue open ... maybe someone will implement this one day :)