2dom/PxMatrix #46: Marquee and scrolling

RobertHolscher, 2018-11-03:
Hi , do you have an advice how to handle scrolling of text from left to right on the PX matrix library ?
I have tried with the routine to have the wrapping to false and then create a for loop with the SetCursor command going from right to left to - text.length() position. However this does not work.
Do you have an advice not to mess up timing and routines to accomplish this function ?
I know that in the Adafruit GFX there is a ScrollText function but that is giving an error in compiling , not beeing in the library of PXmatrix. Please advice

2dom, 2018-11-03:
Who does your approach not work - what does it do? Someone got it to work along these lines ... https://twitter.com/chellotech22/status/1037969473263652864

RobertHolscher, 2018-11-05:
I have got it working. I have made a procedure driven from void loop() to shift per call a cursor position. This has left the interupt for events in tact.

closed 2018-11-05 by RobertHolscher

2dom, 2018-11-05:
Can you post your code as a reference for others? And maybe a gif :)