2dom/PxMatrix #47: PxMatrix BitBanging

harissutanrafiq, 2018-11-13:
Hi @2dom
Thanks for your awesome library.

I wan a try replace SPI using Bitbanging, i know its slower than SPI but its more easy to implement in enother controller, Arduino or STM32. i have porting the AdaFruitRGB Matrix to STM32
[https://www.youtube.com/watch?v=zSYnSyf723g](https://www.youtube.com/watch?v=zSYnSyf723g)

i believe the code must i change is only this part

#ifdef double_buffer
      SPI.writeBytes(&PxMATRIX_buffer[_display_color][buffer_size*_active_buffer+i*_send_buffer_size],_send_buffer_size);
#else
      SPI.writeBytes(&PxMATRIX_buffer[_display_color][i*_send_buffer_size],_send_buffer_size);
#endif

how i can identified each color on PxMATRIX_buffer ..?

Thanks @2dom

2dom, 2018-11-14:
yes .. you would simply replace the writeBytes call with your custom bitbanging code. There are many examples where people have re-implemented these calls, e.g. for TFT displays.

Changing PxMatrix towards using multiple color lines is not trivial - all internal structures/buffers are crafted so it can be sent to the matrix ‚as is‘, i.e. requiring no manipulation during an update which reduces latency / flicker. Individual pixel information is spread over multiple locations so retrieving individual colors is hard.

Not sure why you would want to do that anyway - PxMatrix is optimized for low bin count. If you want to use parallel color data lines you should rather port another library that operates that way.

2dom, 2018-11-14:
Does the STM32 not have hardware SPI?

harissutanrafiq, 2018-11-14:
Thanks @2dom  for your reply
actualy i dont like jumper between Pin IN and Pin Out when i chain multiple led matrix

> Does the STM32 not have hardware SPI?

Yes STM32 have hardware SPI but they dont have SPI.writeBytes

2dom, 2018-11-14:
Well ... even it STM32 only supports writeByte I would use that rather than bitbang ....

kushalsharma07, 2018-12-08:
Hi 2dom this is kushal Sharma from hindustanmechanosolutions@gmail.com
Can I get library support for red led p10. We are using uno with the current resource.
https://github.com/freetronics/DMD2
And this one with esp 8266 is not stable. It crashes as we start Web server in nodemcu.

closed 2018-12-21 by 2dom