2dom/PxMatrix #53: problems with panels bought in 2018 december

fotosettore, 2018-12-24:
hi
it seems something wrong in "new" panels 64x32
i first series of panels form a seller and no one work fine : random pixel on
pattern test and pixeltime both not working.
i made about 10 building changing cables, panels, d1.mini and nodemcu.
i think that new panel series, after october, change in something  does not make them work with the available pxmatrix library.
i sent a my panels to a user that has a panel that works fine and confirm to me than they do no work.
i saw other users complain about the same problem.
any help ?

2dom, 2018-12-28:
Have you ensured that there is a proper ground connection between ESP and panel? It may also be a timing issue..

fotosettore, 2018-12-28:
yes, i had.
anyhow the ground between power supply of panel and all ground pins of all connector are electrically connected together. so I do not think that's the problem. however i tried to connect ground of power supply to ground of nodemcu. same problem.
the same sketch works fine on 64x64 panels. there are same wiring connection.
please note i'm not only maker that complain about this problem : all last messages of last makers.
the makers before october does not have this trouble.
really very strange.

p.s. i removed some flickering problems using a prototype pcboard made with my cnc. this means that wiring connection must be as short as possible.
[image]

2dom, 2018-12-28:
you could try reducing the SPI speed in the library...

fotosettore, 2018-12-28:
SPI.setFrequency(20000000); 
this one ?

2dom, 2018-12-28:
yes ... that's the one. The 20 MHz might be too much...

fotosettore, 2018-12-28:
I will try and let you know. now I do not have the 32 panel at hand.
many thanks for your support.

fotosettore, 2018-12-28:
tried 
SPI.setFrequency(20000000)
SPI.setFrequency(2000000)
SPI.setFrequency(10000000)
SPI.setFrequency(1000000)
SPI.setFrequency(100000)
SPI.setFrequency(40000000)
SPI.setFrequency(4000000)

unlucky tests.
in SPI.setFrequency(1000000) brightness is low

houk001, 2018-12-29:
I have the same problem with panels 64x32 ordered 11 dec from sryled display
https://www.aliexpress.com/item/P3-RGB-pixel-panel-HD-display-64x32-dot-matrix-p3-smd-rgb-led-module/32728985432.html
2 panels 3 nodemcu, 2 power supplies all the same, even pattern test not work as expected.

fotosettore, 2018-12-29:
this is a half confirm that something changed in new panels :-(

2dom, 2018-12-29:
Ok ... could you upload a picture how it looks like?

fotosettore, 2018-12-29:
this is my "effect"

[image]

2dom, 2018-12-29:
what is it supposed to show?

fotosettore, 2018-12-29:
pixeltime.ino

2dom, 2018-12-29:
which part? The weather icons and the color bars?

houk001, 2018-12-29:
This is my random effect from pattern test Sometimes the upper line is white sometimes yellow but newer is going to the end of panel. Always some parts are black

fotosettore, 2018-12-29:
i take the sketch pixeltime as is, 
i changed only
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
and 
display.begin(16);

2dom, 2018-12-29:
Could you generate a video with the pattern test?

fotosettore, 2018-12-29:
this is the pattern_test.ino with same needed changes

[image]

2dom, 2018-12-29:
I need a video with one full cycle - red,yellow, white

fotosettore, 2018-12-29:
i make video but NO CIRCLE ... all pixels are stopped

fotosettore, 2018-12-29:
in ten minutes i make a video with 64x64 e 64x32.

houk001, 2018-12-29:
For me pixels are moving line by line

 

Now it look like this:

 

https://cloud.hawke.pl/index.php/s/xpap2SQmJkgzaQ5

houk001, 2018-12-29:
There is a movement in your video to. Line after some time go down by one pixel. On my both panels is go the same. 
I waited some time and red line goes to the half of size o panel and go up to the first line again.

Second blue line goes to the bottom of panel and goes up to the half of display again.

No other movement on panel.

houk001, 2018-12-29:
Like this

2dom, 2018-12-29:
@fotosettore ok thanks ... that does not make much sense. 

Just a wild guess that either the latch or the output enable signal hast changed from activt low to active high or vice versa. Lets start with OE - could you try running the pattern test and modifying the latch function to this?

```
void PxMATRIX::latch(uint16_t show_time )
{
  
  digitalWrite(_LATCH_PIN,HIGH);
  digitalWrite(_LATCH_PIN,LOW);
  
  digitalWrite(_OE_PIN,HIGH); 
  delayMicroseconds(show_time);
  digitalWrite(_OE_PIN,LOW);
}
```

fotosettore, 2018-12-29:
of course ... i try in few minutes

houk001, 2018-12-29:
New video uploaded to cloud.

2dom, 2018-12-29:
@houk001 sorry ... cannot watch your videos ... loading forever :(

fotosettore, 2018-12-29:
more lines of led on but does not works 
do you need a pic ?

houk001, 2018-12-29:
Download it and then watch. VLC or similar is working

2dom, 2018-12-29:
ok ... lets try changing the latch pin polarity

```
void PxMATRIX::latch(uint16_t show_time )
{
  
  digitalWrite(_LATCH_PIN,LOW);
  digitalWrite(_LATCH_PIN,HIGH);
  
  digitalWrite(_OE_PIN,LOW); 
  delayMicroseconds(show_time);
  digitalWrite(_OE_PIN,HIGH);
}
```

fotosettore, 2018-12-29:
now ... "half" works :-))
it needs video
please wait

houk001, 2018-12-29:
Some progress :DDD video uploaded

2dom, 2018-12-29:
@houk001 Still cannot watch those videos ... download times 10min+

fotosettore, 2018-12-29:
void PxMATRIX::latch(uint16_t show_time )
{  
  digitalWrite(_LATCH_PIN,LOW);
  digitalWrite(_LATCH_PIN,HIGH);
  
  digitalWrite(_OE_PIN,LOW); 
  delayMicroseconds(show_time);
  digitalWrite(_OE_PIN,HIGH);
}
made this effect:
http://www.homoandroidus.com/github/pxmatrix/VID-20181230-WA0001.mp4

houk001, 2018-12-29:
https://youtu.be/fZrK874I-EU

2dom, 2018-12-29:
Try

```
void PxMATRIX::latch(uint16_t show_time )
{
digitalWrite(_LATCH_PIN,LOW);
digitalWrite(_LATCH_PIN,HIGH);
delayMicroseconds(10);
digitalWrite(_OE_PIN,LOW);
delayMicroseconds(show_time);
digitalWrite(_OE_PIN,HIGH);
delayMicroseconds(10);
}
```

2dom, 2018-12-29:
@houk001 Great ... that works! Very colorful :)

fotosettore, 2018-12-29:
no changes with last void

houk001, 2018-12-29:
Yes, no change with delays

houk001, 2018-12-29:
A step back with first high. No more animated right to left movement.

2dom, 2018-12-29:
ok then ... another day. Cheers!

fotosettore, 2018-12-29:
thanks for your support.
now problems is emerged i hope it will be solved .
have a nice night

houk001, 2018-12-29:
Thanks from me too, i think that we are close to success :D

2dom, 2018-12-30:
Could you try

void PxMATRIX::latch(uint16_t show_time )
{
delayMicroseconds(100);
digitalWrite(_LATCH_PIN,LOW);
digitalWrite(_LATCH_PIN,HIGH);
delayMicroseconds(10);
digitalWrite(_OE_PIN,LOW);
delayMicroseconds(show_time);
digitalWrite(_OE_PIN,HIGH);
delayMicroseconds(10);
}

houk001, 2018-12-30:
Of course, this is result 
https://youtu.be/pdtLYMSmJeY
first line goes only for half of the width but second goes better. on the end display goes black because of power supply 5v 3A that detect overload, I'm waiting for tomorrow to buy more powerful :)

fotosettore, 2018-12-30:
no change for me too.
lines ok but other pixels low brightess
same result of http://www.homoandroidus.com/github/pxmatrix/VID-20181230-WA0001.mp4

bribur, 2019-01-04:
With the latch and OE changes above mine is blank or has a few random lights for about 30 seconds then starts cycling and looks almost right, but after each complete cycle it moves down the page one line and there is ghosting.  I believe I have good grounding.  The E pin connection seems ambiguous, it is not grounded on the board yet is not driven by pxmatrix I think.  I have it floating currently.  Another clue noticed by others is that the chips along the edge changed from FM6124 to FM6126A and the supplier indicates the board is "upgraded".   But looking for FM6126A, a IR remote chip comes up so not sure what the change really is.  Mine has the FM6126A and I got it in December.

2dom, 2019-01-04:
@bribur That comment was very helpful ... I think FM6126A has a bit more logic build in ... will investigate ...

[FM6124.pdf](https://github.com/2dom/PxMatrix/files/2728582/FM6124.pdf)
[FM6126A.pdf](https://github.com/2dom/PxMatrix/files/2728583/FM6126A.pdf)

houk001, 2019-01-04:
I also have FM6126A chip.

2dom, 2019-01-05:
Aparently the 6126A needs tree clock edges during latch ... can someone try this ?

```
void PxMATRIX::latch(uint16_t show_time )
{
  //digitalWrite(_OE_PIN,0); // <<< remove this
  digitalWrite(_LATCH_PIN,HIGH);
  digitalWrite(SPI_BUS_CLK,LOW);
  for (uint8_t latch_count=0; latch_count<3; latch_count++)
  {
    digitalWrite(SPI_BUS_CLK,HIGH);
    delayMicroseconds(1);
    digitalWrite(SPI_BUS_CLK,LOW);
    delayMicroseconds(1);
  }
  digitalWrite(_LATCH_PIN,LOW);
  digitalWrite(_OE_PIN,0); //<<<< insert this
  delayMicroseconds(show_time);
  digitalWrite(_OE_PIN,1);
}
```

houk001, 2019-01-05:
[image]
Only static display after this change, Random effect after reset.

2dom, 2019-01-05:
Reversed latch pin polarity ... can you try again ?

```
void PxMATRIX::latch(uint16_t show_time )
{
  //digitalWrite(_OE_PIN,0); // <<< remove this
  digitalWrite(_LATCH_PIN,LOW);
  digitalWrite(SPI_BUS_CLK,LOW);
  for (uint8_t latch_count=0; latch_count<3; latch_count++)
  {
    digitalWrite(SPI_BUS_CLK,HIGH);
    delayMicroseconds(1);
    digitalWrite(SPI_BUS_CLK,LOW);
    delayMicroseconds(1);
  }
  digitalWrite(_LATCH_PIN,HIGH);
  digitalWrite(_OE_PIN,0); //<<<< insert this
  delayMicroseconds(show_time);
  digitalWrite(_OE_PIN,1);
}
```

houk001, 2019-01-05:
https://youtu.be/3-5ALrJa6JA Now it working like this :D

2dom, 2019-01-05:
can you run the pixeltime example and see how it looks?

houk001, 2019-01-05:
Changed in pixeltime
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
and in setup 
display.begin(16);

Result https://youtu.be/L9G4oel-X-g

houk001, 2019-01-05:
https://pastebin.com/cq45nD9e

houk001, 2019-01-05:
Tried all 3 LINE, ZIGZAG, ZAGGIZ and no success.

2dom, 2019-01-05:
Can you comment out setFastUpdate(true) through the sketch?

houk001, 2019-01-05:
I only find one in setup.
After comment it look like this
[image]

[94 more replies]

closed 2019-01-24 by 2dom