2dom/PxMatrix #57: not printing correctly for 1/8 scan

desiboy17, 2019-01-03:


desiboy17, 2019-01-03:
i am using 32*32 1/8 hub75 led panel with esp32 . I tried all sketchs but ended up with missing pixels.

desiboy17, 2019-01-03:
[image]

desiboy17, 2019-01-03:
this is what i am getting printed.

desiboy17, 2019-01-05:
yes, as my matrix panel has three data lines that is A,B and C.
I have selected 1/8 and also tried others i.e 1/16,1/32,1/4.
but the best result is 1/8 in which some pixels are missing.

desiboy17, 2019-01-05:
Is there any in the library with ESP32 ? Have you tested this with ESP32 ?

desiboy17, 2019-01-05:
also i have tried different scan pattern i.e. line ,zigzag ,zagziz.

2dom, 2019-01-27:
This may be an unsupported pattern ... can you run the pattern_test.ino

desiboy17, 2019-01-28:
this is not a patter ..i have written "Pixel time"
and yes i have tried 'pattern_test.ino' too.

my board has only three data lines A,B and C and resolution is 32*32.

2dom, 2019-02-22:
Yes ...  please run the pattern test sketch from examples and post the result

closed 2019-05-09 by 2dom