2dom/PxMatrix #58: Timing issues on a 32x32 matrix

toblum, 2019-01-06:
Hi Dominic,

I tried to make the script work on a 32x32 matrix with a ESP8266 module. It freezes after a few seconds at different parts of the code.
I found out that I have to set the display_ticker to 0.002 instead of 0.001. Then the animations work, but weather is still freezing after a less than a second. After some more seconds the board resets.
The drawback is also that the image (brightness) is flickering when the timer is set to a higher value.

Did you also experience such behaviour before?

Regards
Tobias

2dom, 2019-01-06:
Did you try setFastUpdate(true)? (See example)

toblum, 2019-01-06:
Hi @2dom,

thank you for your quick reply. Yes, I tried different settings, but it doesn't change the behaviour. No matter how I set fast-update the ESP still crashes when the ticker is set to 0.001 and works but flickers if I set it to 0.002. Weather display crashes in both cases.

Any idea?

Greetings
Tobias

If I put the exception into the decoder I get:
```
Decoding 85 results
0x40240000: tx_pwr_backoff at ?? line ?
0x401004e4: malloc at ?? line ?
0x40258c90: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x401068b0: pvPortMalloc at ?? line ?
0x40231880: esf_buf_alloc at ?? line ?
0x402318a5: esf_buf_alloc at ?? line ?
0x4021f023: ieee80211_getmgtframe at ?? line ?
0x4023de3c: chip_v6_rxmax_ext_dig at ?? line ?
0x40222e8f: ieee80211_send_probereq at ?? line ?
0x4023de58: chip_v6_rxmax_ext at ?? line ?
0x40224e43: scan_remove_probe_ssid at ?? line ?
0x40224a14: scan_start at ?? line ?
0x40224590: ieee80211_regdomain_chan_in_range at ?? line ?
0x4023c850: chip_v6_set_chanfreq at ?? line ?
0x40216ab8: do_memp_malloc_pool at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x401065b5: digitalWrite at ?? line ?
0x40224a4f: scan_start at ?? line ?
0x402083e2: SPIClass::writeBytes(unsigned char const*, unsigned int) at ?? line ?
0x402269bc: chm_start_op at ?? line ?
0x401065b5: digitalWrite at ?? line ?
0x40226980: chm_start_op at ?? line ?
0x40224a04: scan_start at ?? line ?
0x402249d8: scan_start at ?? line ?
0x402249a3: scan_start at ?? line ?
0x40224f11: scan_remove_probe_ssid at ?? line ?
0x4020323c: display_updater() at ?? line ?
0x40235339: ets_timer_handler_isr at ?? line ?
0x40235339: ets_timer_handler_isr at ?? line ?
0x4023537e: ets_timer_handler_isr at ?? line ?
0x40234f7b: ets_snprintf at ?? line ?
0x4010441c: call_user_start_local at ?? line ?
0x40104422: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x40100764: cont_ret at cont.S.o line ?
0x40100705: cont_continue at cont.S.o line ?
0x4023004f: pp_attach at ?? line ?
0x4023009e: pp_attach at ?? line ?
0x402301aa: pp_attach at ?? line ?
0x40100f12: pp_post at ?? line ?
0x4022f157: ppTxPkt at ?? line ?
0x402225cb: ieee80211_output_pbuf at ?? line ?
0x4010499b: wdt_feed at ?? line ?
0x40216371: glue2esp_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 299
0x402166c6: new_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 259
0x40216a5b: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x40216a64: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x4021ca28: etharp_output_to_arp_index at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 768
0x4010351b: lmacRecycleMPDU at ?? line ?
0x4021cc80: etharp_output_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 882
0x4021d62c: ip4_output_if_opt_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 962
0x4010105b: ppCalFrameTimes at ?? line ?
0x4021d6ba: ip4_output_if_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 801
0x4021dd93: ip_chksum_pseudo at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
0x4021a29b: udp_sendto_if_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 463
0x401004e4: malloc at ?? line ?
0x40219c47: pbuf_take_at at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306
0x4021a308: udp_sendto_if at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 463
0x40202f5c: PxMATRIX::drawPixel(short, short, unsigned short) at ?? line ?
0x40106520: millis at ?? line ?
0x4020f05c: Adafruit_GFX::writePixel(short, short, unsigned short) at ?? line ?
0x40208580: Adafruit_GFX::drawChar(short, short, unsigned char, unsigned short, unsigned short, unsigned char) at ?? line ?
0x40208808: Adafruit_GFX::write(unsigned char) at ?? line ?
0x40104aaa: wdt_feed at ?? line ?
0x40204a14: setTime(long) at ?? line ?
0x40104b66: ets_timer_setfn at ?? line ?
0x40104d38: ets_timer_arm_new at ?? line ?
0x401004e4: malloc at ?? line ?
0x40104d38: ets_timer_arm_new at ?? line ?
0x4020b6a0: esp_yield at ?? line ?
0x4020160f: delay at ?? line ?
0x40203f34: setup at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x402029c4: configModeCallback(WiFiManager*) at ?? line ?
0x4020299c: saveConfigCallback() at ?? line ?
0x4020fa78: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
```

toblum, 2019-01-06:
Oh, I just saw I opened the issue in the "wrong" repository. I currently use the https://github.com/2dom/PixelTimes script. I hope it's OK as it's related to pxmatrix.

2dom, 2019-01-16:
Did you try reducing the on-time, e.g. 

`display.display(10);`

toblum, 2019-01-20:
HI @2dom,

thanks for your reply. I could try it again today, but I had no success. The PixelTime script does not make use of the mentioned command. 
I forked the repo and just did some minor changes to make it work with my 32x32 matrix and PlatformIO: https://github.com/toblum/PixelTimes

Unfortunately I couldn't kill the flickering so far.

The strange thing is that the startup animation doesn't flicker and even if I deactivate animation and just display a static image, the flickering is still present.

I think it has to do with the ticker I had to set to 0.002 instead of 0.001, but if I leave it on 0.001, it crashes.

Best regards
Tobias

2dom, 2019-01-27:
`display.display(brightness/30);`

 can be found in the display_updater function ... just replace it with `display.display(10); `

If it works slowly increase that value ...

closed 2019-02-22 by 2dom