2dom/PxMatrix #59: Fullscreen test

ES76, 2019-01-11:
Thanks, your sketch is the first that works.

I would like to test each pixel on a bunch of 32X16 P10 Matrix's, my idea was to fill each pixel with a single color.
display.drawPixelRGB888(0,0,255,0,0); works perfectly, can i fillScreen in a single command?

ES76, 2019-01-11:
Thank you for a great Library i eventually came right with the fillScreen command.

closed 2019-01-11 by ES76

2dom, 2019-01-11:
Just for reference ... PxMatrix references [AdafruitGFX ](https://github.com/adafruit/Adafruit-GFX-Library) an you can therfore use all it's commands.

`display.fillRect(0, 0, 32, 16, 0xFFFF);`

should do the trick

ES76, 2019-01-12:
Awesome, Thanks, i was using the wrong referencing after fillScreen, still new at this.
Thank you for the reply