2dom/PxMatrix #63: _mux_pattern never set default.

DaveDavenport, 2019-01-17:
It looks like _row_pattern is initialized to the default value of _mux_pattern.  (then overwritten in begin).

_mux_pattern is never set. (unless the setter is explicitly called.)

https://github.com/2dom/PxMatrix/blob/3fc6cc05d832f8543e30d34eca8afea1a4689bd3/PxMatrix.h#L208

2dom, 2019-01-20:
Not quite sure if I fully understand. _mux_pattern was not initialized. It is now set to STRAIGHT in begin.

What's the issue with _row_pattern?

DaveDavenport, 2019-01-20:
Row pattern is set to binary:
https://github.com/2dom/PxMatrix/blob/3fc6cc05d832f8543e30d34eca8afea1a4689bd3/PxMatrix.h#L284

But it is the interval right? (and BINARY is enum entry for _mux_pattern).

2dom, 2019-01-20:
Yep ... killed the init ... thanks!

closed 2019-01-20 by 2dom