2dom/PxMatrix #65: P3 64x32 w/FM6126A not displaying first two bytes of row

Hotaman, 2019-01-20:
With the FM6126 init code the panel is running the test pattern code but it won't show the first 16 pixels of a row, they are always off. I thought maybe the first red shift register outputs were bad but swapping red to the end so green is output first proved the first red shift register is working.

If I fill the screen with a color and have display() do the outputting, same results, first 16 pixels of each row are off (just the first 2 bytes of each row buffer). It's like the first shift register that gets hit is brain dead and just shifts the bit through.

I'm having other issues like text not rendering text correctly but since the test pattern is not using any buffers, just outputting directly via SPI I should start debugging here.

It runs through the entire pattern so I know rows are being addressed correctly and the panel is LINE style output.

The SPI speed and Mode don't have any effect except Mode3 drops pixels in the area that is working.
I'm also getting faint artifacts on some rows that should be off,

I'm using a D1 Mini through a 74HCT245 to shift to 5V with E tied to ground. I have tried driving E even though it is not used for row addressing. It is not tied to GND on the panel. I thought it might have something to do with the latching but no love that way either.

Anyone have any ideas?

DaveDavenport, 2019-01-20:
Related to #53 ?

2dom, 2019-01-20:
Yes ... is solved in #53

closed 2019-01-20 by 2dom