2dom/PxMatrix #66: Does it work with HUB08 64*32 ?

Dancopy, 2019-01-21:
This Library works with a Matrix with Protocol HUB08 like this: https://pt.aliexpress.com/item/P4-75-dot-matrix-led-module-3-75mm-high-clear-top1-for-text-display-304-152mm/32605629557.html?spm=a2g0s.13010208.99999999.259.60053c00aEfj2Q
If not, would you have any library for it? Because I can not find any arduino or esp8266 sketch for this module; Thank you very much

2dom, 2019-01-27:
Nope ... not supporting these panels at the moment - sorry.

closed 2019-01-27 by 2dom

CTWF, 2019-04-04:
i use p4. 75 RG color.  64x32.  but limit only 2 panel

Dancopy, 2019-04-04:
> 
> 
> i use p4. 75 RG color. 64x32. but limit only 2 panel

With HUB08 interface?
How do you make the connections? Could you send me your sketch?
Thank you

CTWF, 2019-04-09:
https://twitter.com/chellotech22/status/1078610951312617473?s=19

CTWF, 2019-04-09:
https://www.instagram.com/p/BrkqingnciG/?utm_source=ig_share_sheet&igshid=1ir79wt1gxqow

Dancopy, 2019-04-09:
@CTWF  Could you share the arduino sketch and schematic of connection, please? Thanks

cytrus77, 2020-02-20:
Hi,
@Dancopy , @CTWF could You share the sketch ?
I have similar LED panel and some problems with get it to work.
Thanks in advance.