2dom/PxMatrix #68: Maximum recommended Pixelcount

legatus85, 2019-02-01:
Sorry Guys for posting that here. It is not really an issue but something I'd like to know before spending to much money....:

 I am planning zum Display with that matrices. I am interessted in some experiences of how many displays can reasonable be chained. My first thought was of about 32 * 96 px (meaning 3072 Px). But then I got greedy and would like to now if 128 x 64 px can work, too. And would framerate could be achieved using a ESP8266. 

Thank you for your help in advance.

2dom, 2019-02-21:
Cannot give you an definitive answer since I don't have many panels. I have a 64x64 running just fine so 128x64 should work.  However the library is not built for creating large displays - I would go with one that uses parallel color data lines for that.

legatus85, 2019-02-21:
Thank you for your reply. Your code works simply fine. I found some librarys with parallel output, but those did not work that good. Which one would you choose for the ESP32?

2dom, 2019-02-22:
I would try https://github.com/pixelmatix/SmartMatrix/tree/teensylc

closed 2019-02-22 by 2dom