2dom/PxMatrix #69: chaining issue

underscoreHO, 2019-02-06:
I have two 64x32 P5 panels that are chained. Each Panel works fine when using standalone (correct pattern test). When I chain them, lines 2-3, 6-7, 10-11, ..., are not working. Test pattern runs correct, but does is 3 times starting in the first line, then goes to line 4 and then 3 times the fifth line and so on.

I triple checked the wiring, seems fine. Any idea what could be the reason for that or where I have to look at?

In the code I only change
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
to
PxMATRIX display(128,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
when switching from one to two panels.

2dom, 2019-02-21:
You will have to specify the panel width after display.begin using
`setPanelsWidth(2);`

underscoreHO, 2019-02-21:
Thanks! Had `setPanelsWidth(2);` in there, but with some other lines inbetween. Moving it directly after display.begin did the job.

closed 2019-02-21 by underscoreHO