2dom/PxMatrix #70: Help correct code

ffnull, 2019-02-07:
Im got p10 32x16. im connect all like in manual. My matrix had only A and B. But it doesnt correct work
https://youtu.be/0daGIZM5h6s

2dom, 2019-02-21:
Can you post your code?

ffnull, 2019-02-21:
Im use standart example

ffnull, 2019-02-21:
So, any help?

2dom, 2019-02-22:
Whats the problem with it? Your pattern test seems fine.

It appears to be a 1/4 scan panel so you will have to use `display.begin(4);`

ffnull, 2019-02-22:
Yes. Im try but thre space between images. So its different scan

2dom, 2019-02-22:
Can you post a picture of the problem?

ffnull, 2019-02-22:
Please wait 5 minutes. I need to assembly all

ffnull, 2019-02-22:
https://youtu.be/mkLAldZu10Y see how text Adafruit Matrix shows. There 2
rows that wrong

пт, 22 февр. 2019 г., 12:05 FFNULL DEV <djmorganfreeman@gmail.com>:

2dom, 2019-02-22:
Did you try setting the scan pattern with

```
display.begin(4);
display.setScanPattern(x);
```
where x can be LINE, ZIGZAG, ZAGGIZ, WZAGZIG, VZAG. (see github readme)

ffnull, 2019-02-22:
Yes. Best results was with zigzag. But missed 4 lines

ffnull, 2019-02-23:
Any solution?

ffnull, 2019-02-24:
Please close issue. All is fine. Problem was with broken wire on p10 board. Factory fault.

2dom, 2019-02-24:
Ok.. nevermind

closed 2019-02-24 by 2dom