2dom/PxMatrix #71: Library or hardware pixels limit?

lucianolopesjardim, 2019-02-08:
Hi guys, I got seven panels P10 32x16 1/4 chained totaling 224x16 working seamlessly! My goal is 12 panels to have 384x16, but I'm having issues trying with 8 panels. When I try 256x16 the panels turn black, but when I try 255x16 the panels work but the pixels get disorganized. It looks like the limit is 4091 pixels (4092-1). I'm using esp-wroom-32. Here's the code:

`#define PxMATRIX_MAX_WIDTH 256
#define PxMATRIX_MAX_HEIGHT 16

#include <PxMatrix.h>

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2

// Pins for LED MATRIX
PxMATRIX display(256,16,P_LAT,P_OE,P_A,P_B,P_C,P_D);

void setup() {
Serial.begin(9600);
// Define your display layout here, e.g. 1/8 step
display.begin(4);
display.setPanelsWidth(8);
display.setScanPattern(ZAGGIZ); // ZAGGIZ, ZIGZAG, LINE, WZAGZIG, VZAG
display.setMuxPattern(STRAIGHT); // STRAIGHT, 

display.flushDisplay();

display.setTextColor(display.color565(255, 0, 0));
display.setTextSize(2);
display.setCursor(0,0);
display.print("8 chained panels test");
}

void loop() {
display.display(1);
}`

GilDev, 2019-02-19:
Don’t know about limits but am quite interested in them.

For the pixels disorganization I don’t know, but when you initialize your display (`PxMATRIX display(256,16,P_LAT,P_OE,P_A,P_B,P_C,P_D)`), you are specifying a width of 256. And in the PxMatrix the width and height are defined with 8-bit integers (all constructors are defined like this: `PxMATRIX(uint8_t width, uint8_t height,uint8_t LATCH, uint8_t OE, uint8_t A,uint8_t B, …)`), so when you go over 255 it will loop over to 0 and will mess up your config.

I don’t see why it wouldn’t work with a 16-bit integer, you could try and update the code to try it, then maybe make a pull request as it could be interesting to more people and I guess going from 8 to 16 bits wouldn’t bring downsides as ESPs have 32-bit cores.

2dom, 2019-02-21:
Good point ... pushed this change to the latest master. Let me know if it does the trick ..

GilDev, 2019-02-21:
For reference: commit 7843e57bd78f7e66c346d4c94838ed1634d14342

2dom, 2019-02-21:
Can you post a picture of that monster panel if it works ... that would be rather interesting :)

closed 2019-05-09 by 2dom