2dom/PxMatrix #72: P5 64x32 power consumption

GilDev, 2019-02-10:
Hello,

I’m sorry to post this here, but I wanted to check something regarding power consumption.
I’ve got a P5 64x32 display from SRYLED, which works fine with an ESP32 and PxMatrix. Sadly, when I set all LEDs to white at full brightness (0xFFFF color), it consumes about 1.5 A of current and isn’t *that* bright. Is that normal? I thought it could take up from 5 to 8 A for this resolution.

Thanks!

GilDev, 2019-02-15:
Got a response from SRYLED who sold me the matrix on AliExpress and they told me the maximum power consumption of that panel is 20 W.
I’ve set the brightness to 0xFF and the color to 0xFFFF, the power supply is not failling and can definitely supply enough current for 20 W, so I don’t know why it won’t light up more…

houk001, 2019-02-17:
Find in code this ant try it to increase
display.display (70);

GilDev, 2019-02-19:
Thanks you for your answer @houk001. I tried to increase it and it is definitely brighter but it gets an unconsistent yellow-tint when all pixels are on (and white). If there is less pixel lit up it’s clear white.

Here are the results of my experiments:

| Draw time, argument of .display() | Prescaler, 2nd argument of timerBegin() | Timing, 2nd argument of timerAlarmWrite() | Current (A) | Comments                      | 
|-----------------------------------|-----------------------------------------|-------------------------------------------|-------------|-------------------------------| 
| 15                                | 80                                      | 2000                                      | 0,56        | Not bright, but clear white   | 
| 15                                | 80                                      | 4000                                      | 0,32        | Not bright, but clear white   | 
| 15                                | 240                                     | 2000                                      | 0,23        | Not bright, but clear white   | 
| 50                                | 80                                      | 2000                                      | 1,16        | Ugly white                    | 
| 60                                | 80                                      | 2000                                      | 1,25        | Ugly white                    | 
| 70                                | 80                                      | 2000                                      | 1,31        | Ugly white                    | 
| 80                                | 80                                      | 2000                                      | 1,39        | Ugly white                    | 
| 90                                | 80                                      | 2000                                      | 1,43        | Very ugly white, slow to draw | 
| 100                               | 80                                      | 2000                                      | N/A         | Boot loop « SW_CPU_RESET »    | 

I don’t fully understand how everything work, with that draw_time and the timing of the interrupt. I just want to display still images/drawings for now, like a fully white background like I’m trying to do now. Is there a way to get even more bright while staying reliable and get true white and not weird colors?

2dom, 2019-02-19:
Please make Sure that you use rather beefy leads to handle the current. The colour tinting you mentioned usually occurs when the supply voltage at the panel drops below 5v (as it happens with thin wires)

GilDev, 2019-02-19:
Yes I just found out about 20 minutes ago that it was because of my alligator clips used to power up the display, I was getting 3V on the board at full brightness. Thanks @2dom, this was indeed the problem. Now I get about 3.5 A when everything is white and I use a draw time of 90.
I'm still not sure how to get the maximum brightness with this combination of timer and draw time, is there a magic formula depending on the application we want? (faster refresh rate vs. brighter LEDs?)

2dom, 2019-02-19:
Yes...it is a trade off between refresh rate and brightness. You may want to try out setFastUpdate(true) for maximum brightness....

GilDev, 2019-02-19:
Thanks you, I will try that tomorrow. Do you know why it is crashing when I go beyond 100 for the draw time? Is there an absolute limit or does it depends on the timer frequency too?
Maybe it does that because it wants to fire a new display interrupt before the previous one is finished? Again, I’m having trouble finding online how the display really works, the multiplexing, the scan rates’ consequences and everything.

2dom, 2019-02-19:
If you increase the draw time beyond a certain value it will spend all time in the display update function which will eventually trigger the watchdog. I wrote a few lines in the readme how they work...not that complex really...

GilDev, 2019-02-19:
Alright that’s what I thought, so the watchdog is enabled by default.
So I guess we can specify a long draw time and fewer interrupts or a short draw time and more interrupts and get about the same result (in brightness)?

And yeah thanks for your explanations, it’s just that I can’t find schematics and those PR4538 drivers’ datasheets are all in chinese… 😄 I’m just having trouble understanding what is the impact of changing draw time (`show_time` in the `display()` method), it seems to be the time a line is lit up in microseconds.
I guess I need to look into using both cores of the ESP32 so that one can be dedicated to drawing to and updating the display and the other for handling network connections and everything else.

closed 2019-02-21 by 2dom