2dom/PxMatrix #73: 64 x 64 FM6126A panel not working

fnascorral, 2019-02-13:
I am trying very hard to make work a 64 x 64 panel with a FM6126A register shifter, unfortunately without success. I see you updated this library to support this chip.
I am using an ESP32 dev board, the panel is connected per instruction to the ESP32, but using 74HCT245 as voltage level shifters (without them the panel stays black).
Trying the pixeltime example.
I set width and height to 64. setScanPattern and setMuxPattern to default.
display.begin(32) and setDriverChip(FM6126A)
I am running in some issues.
With these setting the ESP32 crashes because a triggered watchdog.
if I change the timerAlarmWrite value (line 99 on pixeltime.ino) from 2000 to a large value like 3000 I don't get the crash anymore but the led matrix shows just some random color lines. The panel is properly grounded. I tested two different same model panels with same results.

Any suggestions?

fotosettore, 2019-02-15:
go here 
[(https://github.com/2dom/PxMatrix/issues/53)]
dominic is working about it ...

fnascorral, 2019-02-15:
Glad to hear... be glad to help with testing or anything else is needed

fnascorral, 2019-02-16:
Ok I did some more testing and checked the signals with a scope:

Pattern test:
red: skips first 16 led, then ok
Yellow: first 16 led display green, then yellow completes ok
White: remains yellow

On the scope I watched CLK and pin R1 and I can see clock signals and data getting clocked on R1. Sorry I don't have a logic analyzer to verify proper timings.

pixtime:

displays two unchanging red squares.
On the scope I do not see any clock signal or data getting clocked on R1. ESP32 is not crashed or at least I don't see a crash dump on the serial port

2dom, 2019-02-22:
The crashing has todo with to much time spent in the display update function - just use larger values as you did. Regarding the random colors - try properly resetting the entire set-up a few times (power off, press reset on the ESP). Make sure the panel is powered when the ESP is booting up. 

We send some configuration data to the FM6126A chip at start-up and it seems to very picky when it accepts it ...

fnascorral, 2019-02-23:
> The crashing has todo with to much time spent in the display update function - just use larger values as you did. Regarding the random colors - try properly resetting the entire set-up a few times [...]

Tried that. What I don't understand is with the same setup: ESP32 connected to the panel HUB75 as per instructions but through 74HCT245 level shifters if I run the pattern test I clearly see the clock signal and R1 data being clocked on the oscilloscope. If I run the pixtime sample there is no clock signal on the scope nor data sent to R1. This is with the time increased to 3000 and no obvious crash.

2dom, 2019-05-09:
What time increased to 3000, show time? That won't work.

kkaoo, 2019-06-27:
ESP32 & FM6126A latch signed,  You can use bit transfer:
SPI.transferBits(v, NULL, 5);
digitalWrite(_LATCH_PIN,HIGH);
SPI.transferBits(v>>5, NULL, 3);
digitalWrite(_LATCH_PIN,LOW);

fnascorral, 2019-06-27:
Thank you kkao your lib works better but still not there. On pixeltime sketch I have a red scrolling text which is better than a black panel. I connected the panel directly so the pins are driven at 3V. This weekend I am going to try to use level shifters and drive the panel pins at 5V and see if it makes a difference.

kkaoo, 2019-07-05:
I haven't tested pixeltime yet, so I'll try it later. I also connect directly to the panel without using level conversion. My board is nodemcu-32s. Excuse me for my poor English.

send my iPhone

发自我的 iPhone

kkaoo, 2019-07-05:
> Thank you kkao your lib works better but still not there. On pixeltime sketch I have a red scrolling text which is better than a black panel. I connected the panel directly so the pins are driven at [...]

I just tested pixeltime sketch, which works fine, but there are two things to note:

"B12a = 0 b0111100011111111; // brightness: low "comment out, because the low brightness setting seems to skew the panel.

I don't have to rewrite setFastUpdate () and setBrightness () related parts, to join shouldn't be hard, you can try to get more perfect results.

2dom, 2019-10-19:
Check out the newest master with ESP32 ... should work now

closed 2019-10-19 by 2dom