2dom/PxMatrix #74: Cannot change real brightness

NatBF, 2019-02-21:
# Issue 
The brightness of [this](https://www.amazon.com/AZERONE-digital-display-2121SMD-256x128mm/dp/B07F2JW8D3/ref=sr_1_1?ie=UTF8&qid=1550705592&sr=8-1&keywords=4mm+pitch+64x32+display) display (sticker says D180463 M1769) does not change despite changing the value of either display.display or display.setBrightness. Both of these values have been changed to anywhere from 0-255 without any effect on the board. (Full code available [here](https://pastebin.com/Qp2amqyN))

# Extra Details
* Used microcontroller is an ESP32
* Board has a resolution of 64x32 with a scan pattern of 16
* [This](https://i.imgur.com/DWdoLQ4.png) sticker reads "D180463 M1769" and is grey 
* [This](https://i.imgur.com/3oXRNlM.png) chip says DP74HC245B 8F0101
* [This](https://i.imgur.com/aBVTtGC.png) chip  says **FM** TC7258EN 5C848100
* [This](https://i.imgur.com/YmAJ8If.png) chip says DP50208 8F0503

2dom, 2019-02-22:
Something was a bit fishy with the dimming code. Can you check out the latest master and run the PixelTime,ino example. I added dimming to the main loop so you should be able to see it right away ...

NatBF, 2019-02-22:
Mamma Mia, that's board's the brightest I've ever seen it :ok_hand:. Thank you for your service, maintaining this library and it's code repository is awesome and great.

NatBF, 2019-02-22:
It should also be noted though that all my scripts stopped functioning until I changed my value "uint8_t display_draw_time=0;" to "uint8_t display_draw_time=10;" so some major legacy code probably got broke somewhere along the line.

2dom, 2019-02-23:
Thanks for the heads up !

closed 2019-02-23 by 2dom