2dom/PxMatrix #75: PxMatrix + DS3231

Albertox0404, 2019-02-22:
Hi, The pins that ESP32 uses are 21 (SDA) and 22 (SCL), I have made a project using one of these screens and a DS3231 but they have pin 22 used by both devices.
For a few minutes both work well being connected on the same pin (22) but in the end they end up showing erroneous information both devices.
I tried to change the STB / LAT pin to one that is not the pin 22 of the ESP32 but the screen does not work with the pins that I have tried.
Can someone help me by telling me which ESP32 pin I can use to connect the STB / LAT other than pin 22?
Or another solution would be to use SoftwareWire with the DS3231 to try to use other pins other than 21 and 22. Anyone who can help with this?

Thank you

2dom, 2019-02-23:
You can assign any pin to the I2C bus hardware, e.g.

```
#define I2C_clk=17;
#define I2C_sda=16;

Wire.begin(I2C_sda, I2C_clk); 
```

Albertox0404, 2019-02-23:
@2dom 
Here is my code, I've put Wire.begin (I2C_sda, I2C_clk); in the SETUP but it still does not work, only the DS3231 works when I connect it to pins 21 (SDA) and 22 (SCL), this is what it returns in the Serial port when it works.

02/23/2019 13:41:34
 24.25C

And this is what it returns in the Serial port when I change the pins.

RTC lost confidence in the DateTime!
85/165/2009 37:165:
-0.25C

`
// CONNECTIONS:
// DS3231 SDA --> SDA
// DS3231 SCL --> SCL
// DS3231 VCC --> 3.3v or 5v
// DS3231 GND --> GND

/* for software wire use below
#include <SoftwareWire.h>  // must be included here so that Arduino library object file references work
#include <RtcDS3231.h>

SoftwareWire myWire(SDA, SCL);
RtcDS3231<SoftwareWire> Rtc(myWire);
 for software wire use above */

#define I2C_clk 17
#define I2C_sda 16

/* for normal hardware wire use below */
#include <Wire.h> // must be included here so that Arduino library object file references work
#include <RtcDS3231.h>
RtcDS3231<TwoWire> Rtc(Wire);
/* for normal hardware wire use above */

void setup () 
{
    Serial.begin(57600);

    Serial.print("compiled: ");
    Serial.print(__DATE__);
    Serial.println(__TIME__);

    //--------RTC SETUP ------------
    // if you are using ESP-01 then uncomment the line below to reset the pins to
    // the available pins for SDA, SCL
    

    Wire.begin(I2C_sda, I2C_clk);

    //Wire.begin(21, 0); // due to limited pins, use pin 0 and 2 for SDA, SCL
    
    Rtc.Begin();

    RtcDateTime compiled = RtcDateTime(__DATE__, __TIME__);
    printDateTime(compiled);
    Serial.println();

    if (!Rtc.IsDateTimeValid()) 
    {
        // Common Cuases:
        //    1) first time you ran and the device wasn't running yet
        //    2) the battery on the device is low or even missing

        Serial.println("RTC lost confidence in the DateTime!");

        // following line sets the RTC to the date & time this sketch was compiled
        // it will also reset the valid flag internally unless the Rtc device is
        // having an issue

        Rtc.SetDateTime(compiled);
    }

    if (!Rtc.GetIsRunning())
    {
        Serial.println("RTC was not actively running, starting now");
        Rtc.SetIsRunning(true);
    }

    RtcDateTime now = Rtc.GetDateTime();
    if (now < compiled) 
    {
        Serial.println("RTC is older than compile time!  (Updating DateTime)");
        Rtc.SetDateTime(compiled);
    }
    else if (now > compiled) 
    {
        Serial.println("RTC is newer than compile time. (this is expected)");
    }
    else if (now == compiled) 
    {
        Serial.println("RTC is the same as compile time! (not expected but all is fine)");
    }

    // never assume the Rtc was last configured by you, so
    // just clear them to your needed state
    Rtc.Enable32kHzPin(false);
    Rtc.SetSquareWavePin(DS3231SquareWavePin_ModeNone); 
}

void loop () 
{
    if (!Rtc.IsDateTimeValid()) 
    {
        // Common Cuases:
        //    1) the battery on the device is low or even missing and the power line was disconnected
        Serial.println("RTC lost confidence in the DateTime!");
    }

    RtcDateTime now = Rtc.GetDateTime();
    printDateTime(now);
    Serial.println();

	RtcTemperature temp = Rtc.GetTemperature();
	temp.Print(Serial);
	// you may also get the temperature as a float and print it
    // Serial.print(temp.AsFloatDegC());
    Serial.println("C");

    delay(1000); // ten seconds
}

#define countof(a) (sizeof(a) / sizeof(a[0]))

void printDateTime(const RtcDateTime& dt)
{
    char datestring[20];

    snprintf_P(datestring, 
            countof(datestring),
            PSTR("%02u/%02u/%04u %02u:%02u:%02u"),
            dt.Month(),
            dt.Day(),
            dt.Year(),
            dt.Hour(),
            dt.Minute(),
            dt.Second() );
    Serial.print(datestring);
}
`

closed 2019-04-19 by 2dom