2dom/PxMatrix #79: Fill complete buffer of matrix and switch immediately on

RobertHolscher, 2019-03-17:


RobertHolscher, 2019-03-17:
I am having a 64x32 RGB matrix and an ESP8266. I am currently uploading images to my ESP over wifi 64 pixels in HEX per row. This because the left over buffer is too small to handle all communication at ones and store it in RAM or ROM. I am writing per line but would like to know if I can write to the matrix board per line, but show the image with one command (so filling the matrix registers first and then switch them on with the loaded image.
Is that possible ? What command should I use to get this effect ?

2dom, 2019-04-09:
You would probably use double buffering ... write to one buffer while displaying the other. It is implemented but has not been tested yet. Wann give it a try?

closed 2019-04-19 by 2dom