2dom/PxMatrix #80: ESP8266 only works when passing th sketch, but not in "normal" mode

sergiocasero, 2019-03-20:
Hello!

First of all, thanks for the library!!

I'm facing strange issue. If I powern on the ESP && screen, nothing happens, but if I connect it to the PC and pass the sketch it works perfectly. I can disconnect the cable without any problems, but if switch power off and on again, it occurs again, I only can run it by passing the sketch.

Any idea?? Thanks!

khawajamechatronics, 2019-03-21:
Did you test by just connected to power and press reset button?

sergiocasero, 2019-03-21:
Yep, it's really really really strange. 

This is the sketch I'm using: https://github.com/sergiocasero/instagram_counter/blob/master/instagram_counter_matrix.ino

I suspect that could be an issue with my specific board, I'll need to test with another one.

PD: I've also tested by disconnecting the P_LAT (because is the D0 GPIO) but it still not working.

2dom, 2019-04-19:
Do you use a proper power supply with beefy cables (5V at board).

sergiocasero, 2019-04-19:
Yep, the power supply I'm using -> https://www.amazon.es/dp/B01HRR9GY4/ref=cm_sw_r_tw_dp_U_x_hJHUCbGCX369D

2dom, 2019-04-19:
Could You measure the voltage at the panel / esp during operation. Also try removing the programming cable and power the esp from the power supply....

PeterTeunissen, 2019-04-22:
The esp+display is very sensitive to proper powersupply. I used https://smile.amazon.com/gp/product/B07CMM2BBR/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1 which I returned even after adding some capacitors to smooth things out, the display was still glitchy.

The one I currently use works great (but the case is pretty industrial): https://smile.amazon.com/gp/product/B0076UJZX0/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

closed 2019-05-09 by 2dom