2dom/PxMatrix #83: Seems don't work

federicoprimomo, 2019-04-08:
Hi, pattern test does not look good, any ideas?
[image]

2dom, 2019-04-09:
Looks perfekt to me ... 1/4 scan with scan pattern ZIGZAG

federicoprimomo, 2019-04-09:
Ok, I set scan pattert to zigzag, but, see this image. It's supposed to display "HOLA"
Thanks.
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("HOLA");
[image]

federicoprimomo, 2019-04-09:
Another example, this image should display a "O"

[image]

GeorgeFlorian, 2019-04-10:
From your pictures it seems that the LED Display is a P10 32x16. That kind of display can have the following Scan Pattern: 1/2, 1/4 or 1/8.
That means 3 different options: `display.begin(2)` , `display.begin(4)` and finally `display.begin(8)`.
Ok ?

So try using `display.begin(2)` first and see what happens. Then try `display.begin(4)`. And then `display.begin(8)`.

If none is working properly try adding `display.setScanPattern(LINE);` and `display.setMuxPattern(BINARY)` to the mix.

Like so: 

```
display.begin(2); // or you can use display.begin(4); or display.begin(8);
display.setScanPattern(LINE); // or try ZIGZAG, ZAGGIZ, WZAGZIG, VZAG
display.setMuxPattern(BINARY); // or you can use display.setMuxPattern(STRAIGHT);
```

Not all displays need `display.setScanPattern();` and `display.setMuxPattern()`. You need to experiment and see what works for you.

Read the documentation and try to understand it, because it's all in there on how to set it up.

closed 2019-05-09 by 2dom