2dom/PxMatrix #84: The colors aren't right ! Video provided.

GeorgeFlorian, 2019-04-09:
Hello !

I am trying to use a LED Display and an ESP32.

Display: [2727 SMD P5 64x32](https://www.aliexpress.com/item/SMD-2727-P5-outdoor-Video-wall-led-module-high-brightness-5500-nits-1-8-scan-64/32788680216.html)
Board: ESP32 Wrover-B DevKitV4

I've done the wiring as it was written in the ReadMe. I thought I did a mistake so I redid the wiring and still nothing.

From the Left HUB75 to the Right HUB75 there are:
R2 to R1
G1 to R2
G2 to G1
B1 to G2
B2 to B1

[image]

I've also redefined the GPIO as follows:
#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 4
#define P_OE 25

P_E should be marked as the GND between G2 and B on the HUB75 Matrix, right ?
I've used E as GPIO 15 together with the proper constructor and the output on the display was chaos. Then I've grounded it and the output was normal. I ended up not using it as a pin on ESP and just grounded it from the HUB75.

I've tried `display.setScanPattern(x)` with all its values: `LINE, ZIGZAG, ZAGGIZ, WZAGZIG, VZAG` and it changes nothing.
I've also used: `display.setMuxPattern(x)` with BINARY and STRAIGHT.
`STRAIGHT` would shift the rows lower and from 8bit x 4 row I would end up with 3 row instead of 4.
Between commenting the function and using `BINARY` there isn't any difference, so I've ended up not using `display.setMuxPattern(x)`.

This is the `Pattern_Test` example code that I am using:

```
#include <PxMatrix.h>

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 4
//#define P_E 15
#define P_OE 25

hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

// Pins for LED MATRIX

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
Serial.begin(115200);
  display.begin(8);
//  display.setScanPattern(LINE);
//  /display.setMuxPattern(BINARY);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

 
   timer = timerBegin(0, 80, true);
   timerAttachInterrupt(timer, &display_updater, true);
   timerAlarmWrite(timer, 2000, true);
   timerAlarmEnable(timer);

  delay(1000);
}

void loop() {

 delay(100);

}
```

Here is a video demonstrating the `pattern_test` :
https://youtu.be/Upmmn8KFX7s

It doesn't look right...

What do you think the problem is ?

2dom, 2019-04-09:
This Looks very wrong, re-check proper power supply (>3A), thick power cables (5V at Panel!), proper grounding between ESP and Panel. If this does not do the trick try reducing the SPI Speed in the PxMatrix.h file.

GeorgeFlorian, 2019-04-09:
> This Looks very wrong, re-check proper power supply (>3A), thick power cables (5V at Panel!), proper grounding between ESP and Panel. If this does not do the trick try reducing the SPI Speed in the PxMatrix.h file.

This is the cable that supplies the LED Display:
[image]

These are the specs of the Power Supply:
[image]

The grounding comes from the Power Supply and the GND pins from the IN - HUB75 matrix (there are 3) and the ESP (I used 1 GND pin) are grounded to that.

### I've noticed a strange behaviour:
You can see in the next video that I have a WHITE jumper cable. That white cable is connected to the GND in-between G2 and B from HUB75 matrix (that should be E) and the GPIO 15 on the ESP32. GPIO 15 is not currently defined in the code, as you can see in the code mentioned in the first post:

```
//#define P_E 15
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
```
One end of the WHITE cable is **always** connected to the GND in-between G2 and B on the HUB75 Matrix.

The other end will be connected to the GPIO15, which is not defined in the code, and then to ground.

You will notice that when it's connected to GPIO15 it works way better, and it has problems only with RED color, and only on the first 2 rows. When the RED starts from the third LED row it works perfectly.
When it's connected to the ground the gibberish happens.
I've changed the `SPI.setFrequency()`  from 10 000 000 to 1 000 000 and it behaves even worse.
It's still bad, but it's bad only for the first 2 rows. I have to point this out.

https://youtu.be/kn8B-_ej34U

Also I've noticed in PxMatrix.h that it uses GPIO 4 `#define SPI_BUS_SS 4`. In my code I've used GPIO 4 for P_D `#define P_D 4`. Is this a problem ?

GeorgeFlorian, 2019-04-09:
```
#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 4
#define P_E 15
#define P_OE 25

#define matrix_width 64
#define matrix_height 32

PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

/**
*
*
*
*/

display.begin(8);
```

I've uploaded `PixelTime example` and this is how it works:
https://youtu.be/5KoXFSEt0MQ

It looks way better through a camera. In reality it isn't that smooth. Reality = 15 fps, camera = 30+ fps.
And the fluctuation between CYAN and MAGENTA can't be seen.
Also the RGB flag is static in reality.

Is that how it should be ?

What if I would like to transform the Display in an entire row ? As I see it now, it prints only on the first row.

GeorgeFlorian, 2019-04-10:
The problems were the wires ! The multi colored wires:
[image]

If you squeeze them tight all the pixel noise disappears and it works perfectly.

I've ended up using E pin aswell.

```
#define P_E 15
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
display.begin(8);
```

closed 2019-04-10 by GeorgeFlorian