2dom/PxMatrix #86: Simple code doesn't work !

GeorgeFlorian, 2019-04-10:
```
#include <PxMatrix.h>

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 4
#define P_E 15
#define P_OE 25

#define matrix_width 64
#define matrix_height 32

hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash

uint8_t display_draw_time=20; //10-50 is usually fine

PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

uint16_t myCOLORS[8]={myRED,myGREEN,myBLUE,myWHITE,myYELLOW,myCYAN,myMAGENTA,myBLACK};

void setup() {
 Serial.begin(115200);
  display.begin(8);
  display.clearDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(16,0);
  display.print("Pixel");
  display.setTextColor(myMAGENTA);
  display.setCursor(13,8);
  display.print("Time");

  delay(3000);
}

void loop() {}
```

The above code doesn't work.
The PixelTime example works fine. But I am trying to display simple, static text. How can I do that ?

fotosettore, 2019-04-11:
esp32 ?

GeorgeFlorian, 2019-04-11:
> esp32 ?
> i tried but lot of problems with esp32
> better using - at moment - esp16 (d1-mini)
> until 60% din ram(50k) is ok, over there are random reset :-(

I figured out how to work with this library, but the bigger problem is that it isn't compatible with ESPAsyncWebServer library.

Reefer to this: https://github.com/2dom/PxMatrix/issues/87

closed 2019-04-12 by GeorgeFlorian