2dom/PxMatrix #87: display_update_enable(true); ->Guru Meditation Error !

GeorgeFlorian, 2019-04-11:
Hello !

I've noticed that I can't properly use this library when using softAP on an ESP32.
I get a lot of Guru Meditation errors.

Deleting `display_update_enable(true);` fixes this, but then I have no way of updating the display.

`display.display()` only flashes the text on screen for a fraction of a second.

Is there any workaround to not be able to use `display_update_enable(true);` ?

NatBF, 2019-04-13:
I know I used to get Guru Meditation Errors too when I would attempt to use the PxMatrix along with SPI, unfortunate but you may have to use another library? Someone made an interface for I2C with these panels and that didn't return errors for them but in my case I just had to have my device process one SPI function then the other or else it would crash because I didn't want to figure out that software.

GeorgeFlorian, 2019-04-14:
> I know I used to get Guru Meditation Errors too when I would attempt to use the PxMatrix along with SPI, unfortunate but you may have to use another library? Someone made an interface for I2C with [...]

So you're saying that the SPIFFS image is actually at conflict with PxMatrix ?

What is the other library you recommend ?

How did you manage to process one function and then the other ?

Thank you !

2dom, 2019-04-14:
If you are rendering static images and use SPIFFS you can always disable the display before read and then enable later. This will be visible as a flicker but won't crash the rsp

GeorgeFlorian, 2019-04-14:
> If you are rendering static images and use SPIFFS you can always disable the display before read and then enable later. This will be visible as a flicker but won't crash the rsp

I am using SPIFFS a lot in my project. I write and read stuff inside txt files. Also, all of the html, css and javascript files are mounted using SPIFFS with `ESPAsyncWebServer.h`:

```
server.on("/html", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/test_file.html", "text/html");
  });
```

I also have a HTTP client inside `void loop()` that's constantly checking for new values from a URL. The values that come from the URL are displayed on the LED Display.

GeorgeFlorian, 2019-04-16:
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40173ab0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'edGuru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1�@t��?d (Double exception)
Guru Meditation Error: Core  1�@t��?d (Unhandled debug exception)
Debu�@6
 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6
 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX [... cut ...] u�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1�����?���?PX�?Debu�@6p��?���?PX�?Debu�@6Debu�@6% 1����
```

This isn't normal, right ?

GeorgeFlorian, 2019-04-16:
This is a sample of my project.
Trying to acces `http://*myIP*` or `http://*myIP*/url` sometimes work, sometimes it returns the above mentioned Guru Meditation Error.

```
#include <Arduino.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>
#include <HTTPClient.h>
#include <PxMatrix.h>

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 4
#define P_E 15
#define P_OE 25
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#define matrix_width 64
#define matrix_height 32

PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

uint8_t display_draw_time= 40;

uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}

void display_update_enable(bool is_enable){
  if (is_enable) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  }
  else  {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
}

void fileReadLines(File file, String x[]) {
  int i = 0;
    while(file.available()){
        String line= file.readStringUntil('\n');
        line.trim();
        x[i] = line;
        i++;
        Serial.println(line);
    }    
}

unsigned int number = 0;
String IPAddr = "";
[... 143 lines cut ...]
      http.begin(URL); //Specify the URL
      urlRead.close();
      display_update_enable(true);
      int httpCode = http.GET(); //Make the request
  
      if (httpCode > 0) { //Check for the returning code
        Serial.println((String)"[HTTP] GET... code: " + httpCode);
        if(httpCode == HTTP_CODE_OK) {
          String payload = http.getString();
          Serial.println(payload);
          parse_URL(payload);
        }
      } else {
          Serial.println((String)"[HTTP] GET... failed, error: " + http.errorToString(httpCode).c_str());
        }  
      http.end(); //Free the resources
      
      } else {
        Serial.println("URL Link is invalid ! Please enter another URL");
      }
  }
  show_on_display(number);
  delay(3000);
}
```

NatBF, 2019-04-17:
Is there some way you can get a boolean state of whether your arduino is currently interacting with the network at all? If so you could do something like
```
void display_update_enable(bool is_enable){ 
  if (is_enable) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  }
  else  {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
}
```
into
```
void display_update_enable(bool is_enable){ 
  if (is_enable && !checkinternetuse) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  }
  else  {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
}
```

then that might fix the feeling that you're "on thin ice" whenever you're interacting with it but I'm just spitballing

2dom, 2019-04-19:
Yeah ... network / wifi activity, SPIFFS etc. and PxMatrix don't go too well together at the moment. I suspect that this would require some investigation into thread-safeness of the whole thing. I won't have the time to do this anytime soon so maybe one of you guys want to give it a shot`?

GeorgeFlorian, 2019-04-19:
> Yeah ... network / wifi activity, SPIFFS etc. and PxMatrix don't go too well together at the moment. I suspect that this would require some investigation into thread-safeness of the whole thing. I won't have the time to do this anytime soon so maybe one of you guys want to give it a shot`?

I would love to, but I seriously lack the skills.

closed 2019-05-09 by 2dom

bribur, 2019-05-16:
Hi:
I realized this is closed because of major surgery involved in fixing it but there is a workaround that almost works, that is to use the Ticker display timer intended for esp8266 instead of the esp32 interrupt approach.  I have wifi, BLE and SPIFFS all working that way.  The problem is the display flickers, so I am wondering if there is some simpler bandaid to make Ticker/Pxmatrix and esp32 play together.  I am able to turn off WIFI and only connect when needed to update a clock but BLE does not turn off/on cleanly yet.  The display is stable when they are disconnected though.  Any thoughts?

GeorgeFlorian, 2019-05-16:
> Hi:
> I realized this is closed because of major surgery involved in fixing it but there is a workaround that almost works, that is to use the Ticker display timer intended for esp8266 instead of the [...]

You should make a separate issue.
I dropped PxMatrix in favor of SmartMatrix, because it's faster, uses DMA, it has a simpler wiring and works perfectly with my project.

bribur, 2019-05-16:
Thanks, I'll look at it.

Nischal-Poudel, 2022-04-17:
how did you solve the issue?, I am having the same issue now

bribur, 2022-04-18:
I moved to Smartmatrix but that requires different wiring if I recall
correctly, it's been a few years.  I eventually abandoned that project and
have had better luck with an RPI zero w, there are boards available and it
just seems more robust.  I'm not sure whether it will do wifi and bluetooth
together though.

Nischal-Poudel, 2022-04-18:
Hey thank you soo much for replying, since I'm working and facing the same
issue, can I get more peak into your project?? And how did you complete
your project?? Knowing that would be soo much useful for me.