2dom/PxMatrix #89: Where I must put the "display.setDriverChip(FM6126A);"

Robiv8, 2019-04-16:
Hallo,
i have a P3 64x32 with the FM6126A Chip and want build the Clock.
Where I must put the code in the Sketch "display.setDriverChip(FM6126A);"
I have try it here:
void setup() {
  Serial.begin(115200);
  display.begin(16);
  setDriverChip(FM6126A);
but no lucky!
MorphingClock:92:24: error: 'setDriverChip' was not declared in this scope
   setDriverChip(FM6126A);

maciejmatczak, 2019-04-19:
Hi,

Don't you lack "display." with setDriverChip command?

Also, please use proper code formatting, it really helps.
```c
void setup() {
    Serial.begin(115200);
    display.begin(16);
    display.setDriverChip(FM6126A);
}
```

Robiv8, 2019-04-19:
Thank you, this was the right tip!

closed 2019-04-19 by Robiv8