2dom/PxMatrix #90: Would you be willing to contribute FM6126A and AB panel support to SmartMatrix?

marcmerlin, 2019-04-17:
Howdy,

Your lib is very ingenious: the hack of shifting all 6 colors components serially to save 5 IO pins that are lacking on ESP8266, is cool.
But it's also 6 times slower :) and just not usable if you go to higher resolutions and want an acceptable refresh rate. I personally use 64x96 on my shirt and have to use an ESP32:
http://marc.merlins.org/perso/arduino/post_2019-04-08_Clubbing_-EDM-Festival-and-Burning-Man-LED-Pants-and-Shirt-v4-on-ESP32-with-RGBPanels-and-SmartMatrix_GFX.html

You could of course add ESP32 support to your lib, but SmartMatrix supports it as long as you use the teensylc branch, see 
http://marc.merlins.org/perso/arduino/post_2019-04-01_SmartMatrix_-SmartMatrix-Shield-v4-for-Teensy_-ESP32-shield-with-level-shifter_-and-SmartMatrix_GFX.html#ESP32
for details on how to use ESP32 with that lib.

However, SmartMatrix lacks:
FM6126A support   
AB panel support

If you have spare time on your hands (ha!) and can be interested in upgrading to a better chip than ESP8266, would you be willing to contribute those extra panels to the SmartMatrix code?

2dom, 2019-04-19:
PxMatrix does support ESP32 ....

2dom, 2019-04-19:
No time for working on SmartMatrix - sorry

closed 2019-04-19 by 2dom

marcmerlin, 2019-04-19:
thanks for your reply, I'd consider using your driver, but ESP8266 and the 6 color components over one pin serial shifting are too slow for what I need. I'm going to guess that you can't even scale to 64x64 without heavy flickering due to too low a refresh rate.
Still, I have to give my hat to you, it is a very cool hack to make it work at all on ESP8266 !

2dom, 2019-04-19:
Thanks ... but 64x64 actually works rather well. See the example at the bottom of the Readme.